ROBOTIS-GIT/DynamixelSDK #307: [3.7.0][C++][USB2Dynamixel2][MX-12W] Issue : Moving status at ADDR 46 is wrong?

sigma--, 2019-06-04:
- I'm using `DynamixelSDK ver. 3.7.0`

- I'm using `C++` Language

- I'm using `USB2Dynamixel2` serial converter

- I'm using `MX-12W`

- and I'm having an `issue` like 

  I am oscillating between GOAL positions 100 and 650. I am checking the Moving bit to go to the next command instead of (goal position - present position) like in examples.

The behavior is - it moves to 650, then starts to move to 100 but MOVING bit goes to 0 before 100 is reached and 650 command is issued again. It moves to 650 successfully(MOVING bit is updated correctly during movement), then starts to move to 100... and so on...
NOTE: Adding a 100ms sleep after command didnt help. The motor moved slightly more but the first MOVING bit read was 0.
Code - 

	  <for (int i = 0; i < 10; i++) {
		  //dxl_moving_status = 0;
		  printf("\n\nGoing to position - %d\n", dxl_goal_position[index]);
		  // Write goal position
		  dxl_comm_result = packetHandler->write2ByteTxRx(portHandler, DXL_ID, ADDR_MX_GOAL_POSITION, dxl_goal_position[index], &dxl_error);
		  if (dxl_comm_result != COMM_SUCCESS)
		  {
			  printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
		  }
		  else if (dxl_error != 0)
		  {
			  printf("%s\n", packetHandler->getRxPacketError(dxl_error));
		  }
		  //Sleep(100);
		  do
		  {
			  // Read moving status
			  dxl_comm_result = packetHandler->read1ByteTxRx(portHandler, DXL_ID, ADDR_MX_MOVING_STATUS, &dxl_moving_status, &dxl_error);
			  if (dxl_comm_result != COMM_SUCCESS)
			  {
				  printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
			  }
			  else if (dxl_error != 0)
			  {
				  printf("%s\n", packetHandler->getRxPacketError(dxl_error));
			  }
			  printf("Moving status = %d\n", dxl_moving_status);

		  } while (dxl_moving_status);
		  // Read present position
		  //do
		  {
			  // Read present position
			  dxl_comm_result = packetHandler->read2ByteTxRx(portHandler, DXL_ID, ADDR_MX_PRESENT_POSITION, &dxl_present_position, &dxl_error);
			  if (dxl_comm_result != COMM_SUCCESS)
			  {
				  printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
			  }
			  else if (dxl_error != 0)
			  {
				  printf("%s\n", packetHandler->getRxPacketError(dxl_error));
			  }

			  printf("Reached position:%03d\n", dxl_present_position);
		  } //while ((abs(dxl_goal_position[index] - dxl_present_position) > DXL_MOVING_STATUS_THRESHOLD));

		  // Change goal position
		  if (index == 0)
		  {
			  index = 1;
		  }
		  else
		  {
			  index = 0;
		  }
		}
	  printf("Time taken - %f\n",(float(clock() - begin_time) / CLOCKS_PER_SEC));
  }
[image]

ROBOTIS-zerom, 2019-06-10:
I have tested this issue by modifying the c++read/write example of the Dynamixel SDK (Ubuntu 16.04)
There was no problem.. (The DXL used for testing is old, so not in good condition.)

The "moving" is set to 0 if the speed falls below the threshold.
What is the moving speed value?
I set the moving speed value to 0 (default value).

```
/*******************************************************************************
* Copyright 2017 ROBOTIS CO., LTD.
*
* Licensed under the Apache License, Version 2.0 (the "License");
* you may not use this file except in compliance with the License.
* You may obtain a copy of the License at
*
*     http://www.apache.org/licenses/LICENSE-2.0
*
* Unless required by applicable law or agreed to in writing, software
* distributed under the License is distributed on an "AS IS" BASIS,
* WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
* See the License for the specific language governing permissions and
* limitations under the License.
*******************************************************************************/

/* Author: Ryu Woon Jung (Leon) */

//
// *********     Read and Write Example      *********
//
//
// Available DXL model on this example : All models using Protocol 1.0
// This example is tested with a DXL MX-28, and an USB2DYNAMIXEL
// Be sure that DXL MX properties are already set as %% ID : 1 / Baudnum : 34 (Baudrate : 57600)
//

#if defined(__linux__) || defined(__APPLE__)
#include <fcntl.h>
#include <termios.h>
#define STDIN_FILENO 0
#elif defined(_WIN32) || defined(_WIN64)
#include <conio.h>
#endif

#include <stdlib.h>
#include <stdio.h>

#include "dynamixel_sdk.h"                                  // Uses Dynamixel SDK library

// Control table address
#define ADDR_MX_TORQUE_ENABLE           24                  // Control table address is different in Dynamixel model
#define ADDR_MX_GOAL_POSITION           30
#define ADDR_MX_PRESENT_POSITION        36

// Protocol version
#define PROTOCOL_VERSION                1.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL_ID                          1                   // Dynamixel ID: 1
#define BAUDRATE                        1000000
#define DEVICENAME                      "/dev/ttyUSB0"      // Check which port is being used on your controller
                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      100                 // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      650                // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD     10                  // Dynamixel moving status threshold

#define ESC_ASCII_VALUE                 0x1b

[... 159 lines cut ...]
    else
    {
      index = 0;
    }
  }

  // Disable Dynamixel Torque
  dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_DISABLE, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    printf("%s\n", packetHandler->getRxPacketError(dxl_error));
  }

  // Close port
  portHandler->closePort();

  return 0;
}
```

[image]

sigma--, 2019-06-10:
I set the moving speed value to 0 too.

ROBOTIS-zerom, 2019-06-14:
There is no problem commenting out the getch() function call.
I still do not have a problem.

If you have another Dynamixel, test it with something else.

ROBOTIS-zerom, 2019-07-01:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-07-01 by ROBOTIS-zerom

sigma--, 2019-08-07:
Can you post the logs with the getch() commented out?