ROBOTIS-GIT/DynamixelSDK #314: Trouble with detecting Dynamixel XM430-W350-R

jawad-tech, 2019-06-21:
Hello!!! 
I am using dynamixel XM430-W350-R controlled with OpenCM 9.04 and OpenCM 485 EXP boards. I have followed the Robotis E-Manual for the setup of Arduino to control the dynamixel by OpenCM 9.04  Now the issues are:

1) Whenever I upload the program in the STM of the OpenCM board to check the status of dynamixel 
[image]

2) Also when I use Roboplus manager for detecting servo motor using the dynamixel wizard it cannot detect any servo and keep on searching like this 
[image]

but when i go to robo plus manager and delete the firmware from the board (recovery mode and green light on) the dynamixel wizard detect the servo but there is no specific firmware to update for the servo i used.
[image]

Any solution with the servo not detected???

ROBOTIS-zerom, 2019-07-17:
Hello, Sorry for the late reply.

Dynamixel Wizard does not support X-series dynamixel.
You should use [R+ Manager 2.0](http://emanual.robotis.com/docs/en/software/rplus2/manager/) or [DYNAMIXEL Wizard 2.0](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/)

Please try firmware recovery again with the above software.
And we recommend using U2D2 rather than OpenCM.

jawad-tech, 2019-07-18:
Thanks buddy for your reply. 
1) you mean to say if i will control the dynamixel using my SDK and dynamixel workbench, from my host pc i will replace opencm with U2D2?
2) As far i am concerned, if i will use opencm with dynamixels, and will write my own c++ or python code, then is that possible? What is the possible communication protocol i will use from host pc to opencm for controlling dynamixels? Can i use usb to ttl converter for host pc to open cm uart and then this act as a serial communication?

jawad-tech, 2019-07-18:
and one question more If i use arduino on linux to ping or Control dynamixel it works. As i use built in Code in arduino file>examples>opencm9.04>Dynamixel workbench>o_find_dynamixel
my port IS /dev/ttyACM0

but in the dynamixel workbench Package available on github. I Build this Package and then go to examples>build>
./find_dynamixel  /dev/ttyACM0   
it dont find any dynamixel 

./position  /dev/ttyACM0  1000000  2
TxxRxx  incorrect status packetbecause from arduino in linux i know my motor Id IS 2 and baudrate IS 1000000

As i want to know the same dynamixel Workbench in arduino in linux works while the github compiled Package does not work
Why? 
If i want to use the dynamixel workbench or dynamixel sdk from github i have to use U2D2?

OpusK, 2019-07-18:
@jawad-tech 

The Arduino Dynamixel SDK and Workbench examples in OpenCM 9.04 have nothing to do with roboplus or ros package.

Therefore, it can not work together.
The example of the Arduino is that OpenCM9.04 acts as Master.
It does not serve as a bridge between PC and DYNAMIXEL.
If you want to use PC and DYNAMIXEL directly, it is recommended to use U2D2.

> Whenever I upload the program in the STM of the OpenCM board to check the status of dynamixel

If this is the case, check that DYNAMIXEL is powered up properly, that the Baudrate rate is equal to the speed of the sketch code, and that the protocol version is the same.

Please refer to the [e-manual](http://emanual.robotis.com/docs/en/dxl/protocol2/) for basic information about DYNAMIXEL.

closed 2019-08-09 by ROBOTIS-zerom