ROBOTIS-GIT/DynamixelSDK #315: [3.5.4][C++][USB2Dynamixel2][MX106R] Issue : Access violation occurred while writing to certain location

lijing94123, 2019-06-26:
- I'm using `DynamixelSDK ver. 3.5.4 (downloaded as "DynamixelSDK-master")`
- I'm using `C++` Language
- I'm using `USB2Dynamixel2` serial converter
- I'm using `MX-106R`
- and I'm having an `issue`:
- 'Access been denied whenever I am trying to read present position of motors and the program was interrupted and error reported as: Exception: 0xC0000005: An access violation occurred while writing to location 0xxxxxxxxx'

My program is aiming to read and write positions to motors based on data received via udp. And the basic logic of my program is as follows:
-definition of variables
-open port and set baudrate
-initialize motors (enble motors torque)
-initialize udp socket 
-bind to server
-while loop includes: receive data, calculate and write into motors
-close socket
-disable motors torque
-close port

And the program was interrupted when I'm trying to write data into motors or even just read present position of motors and I am wondering how to debug with this issue....

Thank you!

lijing94123, 2019-06-26:
I found that it worked (present position can be read from motors) when I directly changed sample codes under example files in sdk (version3.5.2), but I have included necessary files in project properties...and the error seemed to relate to dxl_x64_cpp.dll, and I am wondering why this happened...

ROBOTIS-zerom, 2019-07-17:
Hello, Lijing.

If the example code of the SDK works properly, it is not a problem with the SDK.
Since an access violation has occurred, perhaps the location of the dxl_x64_cpp.dll file is incorrect.

check the library and .dll file location settings again.

closed 2019-08-09 by ROBOTIS-zerom