ROBOTIS-GIT/DynamixelSDK #318: [3.7.0][C++] CRC Calculation fails to catch errors in packet length

Zabot, 2019-07-03:
I was encountering some issues with stalling motors causing segmentation faults in my C++ code randomly. After some investigation, it appears the CRC of a packet in [Protocol2PacketHandler::rxPacket](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/c%2B%2B/src/dynamixel_sdk/protocol2_packet_handler.cpp#L307-L420) is not computed until after the _entire_ packet has been read.

This can cause buffer overflows if there is a bit error in the length field in such a way that the length is still valid (`< RXPACKET_MAX_LEN`), but larger than the buffer allocated for the packet. `rxPacket` will write past the end of the buffer thinking that there is more to read. The CRC would fail on the resulting packet, but the overflow crashes the program before that point is reached.

ROBOTIS-zerom, 2019-07-18:
I've decreased the size of rxpacket to reduce memory usage, which seems to be a problem.
I will try to find ways to reduce memory usage and solve this problems.

ROBOTIS-zerom, 2020-01-07:
fixed.

closed 2020-01-07 by ROBOTIS-zerom