ROBOTIS-GIT/DynamixelSDK #319: [3.7.0][C++][USB2Dynamixel][XL430-W250-R] Issue : No Status Packet error

suddarsun, 2019-07-03:
- I'm using `**DynamixelSDK ver. 3.7.0`**  

- I'm using `**C++**` Language

- I'm using `**USB2Dynamixel**` serial converter

- I'm using **`XL430-W250`**

  - I am trying to execute the **read_write example** on my Dynamixel XL430 and I get the **[TxRxResult] There is no status packet! error.**

What I have tried so far:

1. Ensured baudrate is set to 57600 in read_write.cpp. This is the default value for the motor as in the e-manual.

2. Ensured the correct addresses are set for torque enable, present position, and goal position

3. Tried changing the latency timer in port_handler_linux.cpp from 16 ms to 8 ms, 3 ms, and 20ms 

After all the above three steps, I still am unsuccessful.

Any help would be appreciated.

Thanks

ROBOTIS-zerom, 2019-07-17:
Hello, suddarsun.

1. Verify that the latency timer is 1ms.
 `$ cat /sys/bus/usb-serial/devices/ttyUSB0/latency_timer`
2. If not, change the latency timer to 1ms
`$ echo 1 | sudo tee /sys/bus/usb-serial/devices/ttyUSB0/latency_timer`
3. Download DYNAMIXEL Wizard 2.0 : http://en.robotis.com/service/downloadpage.php?ca_id=10
4. Test with DYNAMIXEL Wizard 2.0
5. If there is something wrong, try test again after the firmware recovery.

ROBOTIS-zerom, 2019-08-09:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-08-09 by ROBOTIS-zerom

TannerNelson16, 2023-10-14:
I'm having this same issue. Did you ever find the solution? I've tried adjust the latency timer with no effect. Everything works in the dynamixel wizard. I just can't seem to get the test python scripts to work in the DynamixelSDK stuff.