ROBOTIS-GIT/DynamixelSDK #320: [Arduino][OpenCR][DynamixelSDK][MX-28 & MX-106] There is no status packet, groupBulkRead getdata failed

aaceve, 2019-07-04:
I'm using Arduino 1.8.9, DynamixelSDK, C++, OpenCR and some Mx106 & Mx28. I'm trying to use BulkRead for present position (36 in Control Table, 2Bytes width) but I got this message:  

`[TxRxResult] There is no status packet!,   [groupBulkRead getdata failed]`

What I am doing wrong? I have followed the same logic than the one presented on: `Examples > OpenCR > DynamixelSDK > Protocol1.0 > bulk_read.`

I have also read the issue  https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/189,  but I haven't found any clue to solve my problem.

Could someone help me, please? Thanks in advance.

Here's the code I am using:

```
#include <DynamixelSDK.h>
#define DEVICENAME               "OpenCR_DXL_Port"      // Only has a symbolic meaning
#define PROTOCOL_VERSION             1.0                // Protocol version
#define BAUDRATE                    2000000             // Baudrate

void setup() {
  Serial.begin(115200);
  while(!Serial);
  
  dynamixel::PortHandler *portHandler = 
                    dynamixel::PortHandler::getPortHandler(DEVICENAME);
  dynamixel::PacketHandler *packetHandler = 
                    dynamixel::PacketHandler::getPacketHandler(PROTOCOL_VERSION);
  dynamixel::GroupBulkRead groupBulkRead(portHandler, packetHandler);
  int dxl_comm_result = COMM_TX_FAIL;             // Communication result
  uint8_t dxl_error = 0;                          // Dynamixel error
  bool dxl_addparam_result = false;               // addParam result
  bool dxl_getdata_result = false;                // GetParam result
  uint16_t dxl_value = 0;                         // Result of read command
  int PresentPose[19];  // Array containing Present-Values of all connected Dynamixels

  portHandler->openPort();
  portHandler->setBaudRate(BAUDRATE);

    // Preparing Packet      
    for (int dxl_id = 1; dxl_id <= 18; ++dxl_id) {
      dxl_addparam_result = groupBulkRead.addParam(dxl_id, 36, 2);
      if (dxl_addparam_result != true) {
        Serial.print("[ID:"); Serial.print(dxl_id); 
        Serial.println(", groupBulkRead addparam failed]");
      }
    }
    // BulkRead command
    dxl_comm_result = groupBulkRead.txRxPacket();
    delay(1000);  // Seems to be unuseful. Problem persist.
    if (dxl_comm_result != COMM_SUCCESS)
      Serial.println(packetHandler->getTxRxResult(dxl_comm_result));
    else {
      for (int dxl_id = 1; dxl_id <= 18; ++dxl_id) {
        if (groupBulkRead.getError(dxl_id, &dxl_error)) {
          Serial.print("[ID:");Serial.print(dxl_id);Serial.print(",");
          Serial.print(packetHandler->getRxPacketError(dxl_error)); Serial.println("]");
        }
      }
    }
    // Checking if data is available
    for (int dxl_id = 1; dxl_id <= 18; ++dxl_id) {
      dxl_getdata_result = groupBulkRead.isAvailable(dxl_id, 36, 2);
      if (dxl_getdata_result != true) {
        Serial.print("[ID:"); Serial.print(dxl_id); 
        Serial.println(", groupBulkRead getdata failed]");
      }
    }
    // Reading Packet
    for (int dxl_id = 1; dxl_id <= 18; ++dxl_id) {
      dxl_value = groupBulkRead.getData(dxl_id, 36, 2);
      PresentPose[dxl_id] = int(dxl_value);
    }
  

  // Write in monitor the Present-Values of all 18 Mx servos 
  char text2print[100];
  for (int dxl_id = 1; dxl_id <= 18; ++dxl_id) {
    sprintf(text2print," %4d,",PresentPose[dxl_id]);
    Serial.print(text2print);
  }

  portHandler->closePort();
}

void loop() {
}
```

aaceve, 2019-07-05:
By the way, is there a maximum number of servos that can be added to the Bulk package?

In my previous publication, I show a program that asks for the present-value of 18 servos without success, but if I reduce that number to 9, it WORKS! Is that normal behavior? Why?

In fact, it doesn't matter if I ask present-value of servos 1-9 or 10-18 or 6-14, (etc) as long as, they are 9.

ROBOTIS-zerom, 2019-07-17:
Hello, aaceve.

After sending instruction packet, it waits for all dynamixel's status packets for a fixed time(timeout value). However, it seems that this timeout value is not enough to wait for all dynamixel's satus packets to arrive. 

I'll check this tomorrow..

OpusK, 2019-07-18:
Hi, @aaceve 

Did you change the [Return Delay Time](http://emanual.robotis.com/docs/en/dxl/mx/mx-106/#return-delay-time) of DYNAMIXEL?

The problem is that the delay time of the actual packet is larger than the timeout waiting in OpenCR.
This is because the faster the communication speed, the shorter the timeout time in OpenCR.

Change the Return Delay Time and test it.

closed 2019-08-09 by ROBOTIS-zerom