ROBOTIS-GIT/DynamixelSDK #323: [3.7.0][Python][OpenCR][XM540-W270-R] Issue : "error: (25, 'Inappropriate ioctl for device')" in read_write.py

eolv1, 2019-07-12:
- I'm using `DynamixelSDK ver. 3.7.0`

- I'm using `python` Language

- I'm using `OpenCR serial converter

  

- I'm using `XM540-W270-R`

- and I'm having an issue trying to run the read_write.py in protocol2_0 but I cannot do anything because I keep getting this kind of error 
"error: (25, 'Inappropriate ioctl for device')"

I have no idea what this is. This where it is located in the script.
[image]

Running python 2.7 through Spyder

New to Dynamixel and relatively new to Python, so I appreiate any help that comes my way.

eolv1, 2019-07-12:
Actually tried running on terminal and it works there....have some other erros but working through it. 

Is there a way however to run this on SPyder or an interface that is not the terminal? I understand now that the error has to do with the interface not having certain terminal access.

ROBOTIS-zerom, 2019-07-18:
Sorry for not helping you solve the problem. 
As you know, this is not a problem with the SDK. 
So I want to close this issue. 

You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-07-18 by ROBOTIS-zerom