ROBOTIS-GIT/DynamixelSDK #324: [3.5.4][MATLAB][UD2D][AX-12W] Issue: communication with Matlab failed loadlibrary

salvatoreNRIM, 2019-07-23:
I am using a Dynamixel AX-12W. I am using Matlab 2016b, and USB2Dynamixel and my issue is that I keep getting problem in loading the library in the read_write.m file, attached the error message. Does anybody know how to solve it? 
[image]

asheesh-2019, 2019-09-26:
facing similar issue with AX-18a.

I believe the issue to be in the dynamixelSDK file.

The dynamixelSDK master folder should contain 1 folder and 3 subfolder in " [DynamixelSDK folder]/c/include "

but it cotains only one foder and one subfolder.

issue still unresolved.

johncgh22, 2019-10-03:
Hi @salvatoreNRIM and @asheesh-2019 

I belive this error is because MATLAB doesnt understand some of the commands in the SDK, so it **needs a compiler** in order to build the packages.

I use this compliler for MATLAB:
[https://la.mathworks.com/matlabcentral/fileexchange/52848-matlab-support-for-mingw-w64-c-c-compiler](url)
[image]

Try installing this addon from the link, or in the Add-Ons tab inside MATLAB and give it a try.

asheesh-2019, 2019-10-07:
hi  @johncgh22 

I had installed this compiler previously.

But its not working.

ROBOTIS-Will, 2020-09-17:
This issue thread will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-09-17 by ROBOTIS-Will