ROBOTIS-GIT/DynamixelSDK #326: [3.7.0][python][OpenCR][XM540-W270-R] Issue : incorrect status packet/no status packet error when running ping.py`

eolv1, 2019-07-26:
Hello everyone, 
- I'm using `DynamixelSDK ver. 3.7.0`
- I'm using `python` Language
- I'm using `OpenCR` serial converter
- I'm using `XM540-W270-R`
-Using protocol2_0 as per the control table of the motor
- and I'm having an `no status packet/incorrect status packet` when trying to run ping.py in tests

I have tried:
checking the addresses and values of the inputs match the motor control table
ive changed the baudrate and tried all possible values (when baud was 4.5M I got an error for the baud so I went back to 57600)
changing the protocol version just because 
connecting/disconnecting hardware and restarting
i have the correct serial port

when i run ping.py this is what I get 
[image]

this is the code...devicename is correct so its not dev/tty

import os

if os.name == 'nt':
    import msvcrt
    def getch():
        return msvcrt.getch().decode()
else:
    import sys, tty, termios
    fd = sys.stdin.fileno()
    old_settings = termios.tcgetattr(fd)
    def getch():
        try:
            tty.setraw(sys.stdin.fileno())
            ch = sys.stdin.read(1)
        finally:
            termios.tcsetattr(fd, termios.TCSADRAIN, old_settings)
        return ch

from dynamixel_sdk import *                 # Uses Dynamixel SDK library

# Protocol version
PROTOCOL_VERSION        = 2.0               # See which protocol version is used in the Dynamixel

# Default setting
DXL_ID                  = 1              # Dynamixel ID : 1
BAUDRATE                = 57600            # Dynamixel default baudrate : 57600
DEVICENAME              = '/dev/ttyACM0'    # Check which port is being used on your controller
                                            # ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

# Initialize PortHandler instance
# Set the port path
# Get methods and members of PortHandlerLinux or PortHandlerWindows
portHandler = PortHandler(DEVICENAME)

# Initialize PacketHandler instance
# Set the protocol version
# Get methods and members of Protocol1PacketHandler or Protocol2PacketHandler
packetHandler = PacketHandler(PROTOCOL_VERSION)

# Open port
if portHandler.openPort():
    print("Succeeded to open the port")
else:
    print("Failed to open the port")
    print("Press any key to terminate...")
    getch()
    quit()

# Set port baudrate
if portHandler.setBaudRate(BAUDRATE):
    print("Succeeded to change the baudrate")
else:
    print("Failed to change the baudrate")
    print("Press any key to terminate...")
    getch()
    quit()

# Try to ping the Dynamixel
# Get Dynamixel model number
dxl_model_number, dxl_comm_result, dxl_error = packetHandler.ping(portHandler, DXL_ID)
if dxl_comm_result != COMM_SUCCESS:
    print("%s" % packetHandler.getTxRxResult(dxl_comm_result))
elif dxl_error != 0:
    print("%s" % packetHandler.getRxPacketError(dxl_error))
else:
    print("[ID:%03d] ping Succeeded. Dynamixel model number : %d" % (DXL_ID, dxl_model_number))

# Close port
portHandler.closePort()

Any help is appreciated! 
I feel like I've tried everything on the boards I found on this issue

eolv1, 2019-07-26:
the problem seems to start under 'get dynamixel model number' 

I have opened the protocol2_packet_handler.py in build->lib and the same error occurs

ROBOTIS-zerom, 2020-01-08:
What firmware did you use with OpenCR?
Can you touch DYNAMIXEL using the DYNAMIXEL Wizard 2.0?

We  recommend USB2DYNAMIXEL or U2D2 as interface device.
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/device_setup/

closed 2020-01-21 by ROBOTIS-zerom