ROBOTIS-GIT/DynamixelSDK #328: unable to install dynamixel sdk

aswinkumar2019, 2019-08-01:
root@aswin-desktop:~/Drivers/DynamixelSDK/c++/build/linux64# make
mkdir -p ./.objects/
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_bulk_read.cpp -o .objects/group_bulk_read.o
g++: error: unrecognized command line option ‘-m64’
Makefile:117: recipe for target '.objects/group_bulk_read.o' failed
make: *** [.objects/group_bulk_read.o] Error 1

Please help me to fix this error

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1], [3.5.2], etc.

- I'm using `C++` Language

  - [C] / [C++] / [C#] / [Python] / [Java] / [MATLAB] / [LABVIEW] / [ROS] / [Arduino]

- I'm using `USB2Dynamixel2` serial converter

  - [USB2Dynamixel] / [USB2Dynamixel2] or [OpenCM904] / [OpenCR] / [CM700s] / [CM500s] or other ROBOTIS product only

- I'm using `XM430-W210-R`

  - [MX-28], [AX-12W], [H54-200W], etc.

- and I'm having an `issue` like what

  - 'My motor doesn't work', etc.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

aswinkumar2019, 2019-08-03:
Is the project active????

ROBOTIS-zerom, 2019-08-12:
Hello,

Make sure you are using a 64-bit OS.
Perhaps your g++ compiler is for 32bit only.

aswinkumar2019, 2019-08-12:
sorry,i think i checkedout kinetic-devel while installing ros melodic.I think this may be the reason for the issue.I shall check it out and update response here

aswinkumar2019, 2019-08-12:
I dont think these two are the issues,i have 64bit OS

ROBOTIS-zerom, 2019-08-12:
A simple way to solve the problem is to simply remove the -m64 flag.

Are you sure you have an x86_64 processor?
Because -m64 is only valid for x86_64 processors..
It cannot be used on processors like arm64.

aswinkumar2019, 2019-08-13:
Sorry, I have jetson nano which is based on arm64. I am not sure.
I can't use dynamixel sdk with jetson nano or arm64 devices????

ROBOTIS-zerom, 2019-08-13:
No.. you just remove the -m64 flag in the Makefile.

Change below line
```
FORMAT     = -m64
```
to..
```
FORMAT     =
```

aswinkumar2019, 2019-08-13:
It works perfectly, thank you very much

closed 2019-08-13 by ROBOTIS-zerom