ROBOTIS-GIT/DynamixelSDK #329: [3.7.0][C][C#][C++][LN-101][XL-320] Issue : Compile errors in all platforms

uwouldknow, 2019-08-01:
This is what I am using:
Windows 10
DynamixelSDK version 3.7.0
Visual Studio 2019
[C] / [C++] / [C#]
xl-320
opencm9.04
LN-101

Here is my issue:

Hello I am trying to get the SDK to work with c# so I can use it with unity but I am running into errors.... Firstly as soon as I open the C# solution "protocol_combined.sln" I recieve these errors shown below..... 

[image]

I was able to change to code so that it would at least compile as shown below......

[image]

After this I ran the code and received this error shown below......

[image]

Next I tried getting C and C++ to work but ran into issues as well.... with both languages I ran into the same error when running the "protocol_combined.sln" that is shown below....

[image]

Also am I supposed to be doing something special when the option in the picture below shows up?

[image]

Furthermore do I have to use Visual Studios 2015 or can I use the new version of Visual Studios and if I use the new version I would have to change that setting in the picture below right? Same question with the WindowsSDK version as well

[image]

And yes I followed all of the steps from this website http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/overview/#dynamixel-sdk

Overall I just want it to work in c# but I thought I should mention my issues with c and c++ as well.

colejbutler, 2019-08-03:
Update: I recently tried with visual studio 2015 express and received the same error

ROBOTIS-zerom, 2019-08-13:
OK. I found a bug in the code.
I will fix the bug and upload it, and notify you again.

ROBOTIS-zerom, 2019-08-13:
I fixed the bug. Check out the latest master branch..

closed 2019-10-29 by ROBOTIS-zerom