ROBOTIS-GIT/DynamixelSDK #331: Dynamixel SyncRead issues

ksmola, 2019-08-09:
Hi, 

I'm using two H54P-200-S500-R Dynamixel Pro, Firmware 10.
C application on Linux, native RS485 port.

Everything works as it should during a SyncRead as long as the Dynamixels are configured as ID 1 and 2. 
If I configure them as ID 2 and 3, I get no response to a SyncRead.
If configured as ID1 and ID3, I only get a response from ID1. (confirmed this with a logic analyzer)

Is it expected that they only respond to a SyncRead when configured starting at ID1 + only consecutive IDs?

ROBOTIS-zerom, 2019-08-12:
Have you tested using this example?
DynamixelSDK/c/example/protocol2.0/sync_read_write/sync_read_write.c 

Did you change the value of DXL1_ID and DXL2_ID and compile them?

ksmola, 2019-08-13:
Thanks! I haven't tried that sample yet, but I am sending the correct sync read packet out, it's broadcast (0xFE), so ID independent.
Shouldn't all Dynamixels on the bus respond no matter what ID they are set to?
Could this be an issue with Firmware 10 (FW 11 is available, I couldn't find a changelog online)
I confirmed with a scope and logic analyzer that the issue is not on the client app reading the value, the Dynamixel simply never responds.

ROBOTIS-zerom, 2019-08-13:
http://emanual.robotis.com/docs/en/dxl/protocol2/#sync-read

See the example packet.
packet's ID is 0xFE(broadcast) but look at the parameters.. 
Parameter 4+1 : ID of the 1st Device
Parameter 4+2 : ID of the 2nd Device

The firmware version does not matter..
See the link below for how to check the firmware version information.
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#firmware-information

ksmola, 2019-08-13:
Thank you so much, I think this solves it.

ksmola, 2019-08-13:
I checked the message that's being sent out, and it's indeed correct:
`FF FF FD 00 FE 0A 00 82 44 02 04 00 01 02 03 29 D0`

It seems like the issue is that I'm querying ID1 to IDn (n being the highest ID on the bus, so I also query IDs that are not there)
In the example above, only ID1 and ID3 are connected, but I also query for ID2s position and only ID1 replied.
Would that cause a problem?

ROBOTIS-zerom, 2019-08-13:
You must include only the ID of the device to read from.
If any of the IDs included in the parameter do not respond, the instruction packet will fail.

ksmola, 2019-08-14:
Sounds good, I'll add some functionality to scan the bus and identify every servo -- just wanted to make sure this is expected behavior. Thank you

closed 2019-08-14 by ksmola