ROBOTIS-GIT/DynamixelSDK #344: Labview- Missing h

abuzaa, 2019-09-08:
ISSUE TEMPLATE ver. 1.1.1

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1], [3.5.2], etc.

- I'm using `C++` Language

  - [C] / [C++] / [C#] / [Python] / [Java] / [MATLAB] / [LABVIEW] / [ROS] / [Arduino]

- I'm using `USB2Dynamixel2` serial converter

  - [USB2Dynamixel] / [USB2Dynamixel2] or [OpenCM904] / [OpenCR] / [CM700s] / [CM500s] or other ROBOTIS product only

- I'm using `XM430-W210-R`

  - [MX-28], [AX-12W], [H54-200W], etc.

- and I'm having an `issue` like what

  - 'My motor doesn't work', etc.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

johncgh22, 2019-09-10:
Hi,

First, check the tutorial for adding the library into LabVIEW:
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/labview_windows/#labview-windows
Here it shows you where are the .dll and .h files you need to build the functions.

I recommend upgrading your SDK, it helps with some problems with communication. Use the latest version, check it here:
https://github.com/ROBOTIS-GIT/DynamixelSDK/releases/tag/3.7.21

abuzaa, 2019-09-10:
Hi @johncgh22 

Thank you for your resposne, but my problem is that I can't find the header file '.h' in the new SDK. Do you have any idea where it is in the zipped folder?
Thanks in advance.

johncgh22, 2019-09-10:
Hi @abuzaa 

For the new SDK, the .h file is locate in this folder:
~~\DynamixelSDK-3.7.21\c\include\dynamixel_sdk
Dont look inside the labview folder in the SDK, since all of the fuctions are made in C, you need to look in the c folder. From there, LabVIEW takes the info and make the VI neccesary. I´ll attach a file for my specific case...

[image]

closed 2020-01-09 by ROBOTIS-zerom