ROBOTIS-GIT/DynamixelSDK #345: [3.7.21][LabVIEW,MATLAB][USB2Dynamixel2][AX-12A] Issue : TxRxResult: There is no status packet!

johncgh22, 2019-09-10:
Hi, and good afternoon everybody: 

- I'm using `**DynamixelSDK ver. 3.7.21**`
- I'm using **LabVIEW 2017**, but I´m also testing on MATLAB 2018a, and 2019a
- I'm using `**USB2Dynamixel**` serial converter, but also I have an U2D2 and a CM-530
- I'm using **`AX-12A`**, three of them to be exact
- and I'm having this message: **[TxRxResult: There is no status packet!**

Let me explain what I´m be dealing for the last week. I´m using 3 AX-12A in a robotic arm, my first test with these motors were made in the new Dynamixel Wizard 2.0 and they worked fine, until yesterday.

I was testing in a custom sync_write.vi I made for using this 3 motors, and suddenly, one of them loose connection. In the VI, it says Communication success, but one of the motors doesn´t move at all.

[image]

I checked them in Dynamixel Wizard 2.0, but the second one keeps sending me: ERROR or TIME OUT. 

[image]
[image]

- All the motors had 1M as a baudrate
- Each one has a different ID
- I´m using a SMPS2Dynamixel and a 12v power supply, but I also have a PowerHub 3P for Dynamixel
- The motor which doesnt move, has its LED on, all the time. I dont know what it means.

I tested them in MATLAB, and I keep getting the same errors. Also i read previous issues and I tested some solutions with no luck:

- Changing baudrate, it doesnt work
- Changing Return Delay Time, it doesn´t work
- I checked, and all the motors had the same configuration.
- I did a Firmware Recovery to that specific Dynamixel, it doesn´t work either
- Using the bulk_read.vi example in LabVIEW/MATLAB, just one of the motors worked, the other one doesn´t move at all.
- I checked the Control Table for an error in the address, but they are Ok
- I tested them with the U2D2 in Dynamixel Wizard 2.0 and 1.0, but it doesn´t work too. Also with the CM-530, same story.

As far as I know, this is the most frecuent issue with Dynamixel, but if someone has another solution/idea why this happen, I will be happy to know. I´ll be adding some screenshots and a video to complement this issue.

[DYNAMIXEL.zip](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/3593142/DYNAMIXEL.zip)

Thanks.

ROBOTIS-zerom, 2019-09-16:
Hi, John.
To clean the overload shutdown, you should reboot the device.

johncgh22, 2019-09-16:
Hi @ROBOTIS-zerom, 

How do I reboot the device? In Dynamixel Wizard 2.0 the option is gray-out and I can´t select it.
[image]

In the SDK, there´s the reboot function, but it´s only available for Protocol 2.0. Is this the method, or how should I proceed?

Thanks in Advance...

ROBOTIS-zerom, 2019-09-16:
You just turn it off and on.
In protocol 1.0, rebooting by communication is not possible.

johncgh22, 2019-09-16:
I´m turning it off and on, but it still sends the same error. Should I be worried about a potencial damage to the servo?

ROBOTIS-zerom, 2019-09-16:
I don't know what you mean by "still send the same error."
The torque will be disabled when the device is turned on. Does it mean that overload shutdown occurs when torque is enabled at no load??

johncgh22, 2019-09-16:
Sorry if I couldn´t explain it correctly.
When I turn on/off the servo, I check it again in Dynamixel Wizard 2.0. After I reconnect it, I select Goal Position (Address 30) to verify if I can control it again, but as I mention before, I still receive the same message "TIME OUT" or just an "ERROR", as the ones I posted in my first post.

The torque is disable when I turn it on, but as soon as I change values in Goal Position, it sends the Overload Shutdown, and all the values in the Wizard resets tp their previous state. The servo is with no load at the moment of testing, and just one motor behaves as I describe.

One note to attach is, when I clic on the Goal Position knob, the Red Light at the servo turns On.

johncgh22, 2019-10-03:
To add an update for this issue, it seems that my motor **is stuck in boot mode.** 
Since I tried many times a Firmware Recovery in Dynamixel Wizard 2.0 with no success, a friend recommend me to Force a "Manual Reboot" using a transistor to recreate a Short Circuit in order to break from it and recover functionality to the motor.

I´ll report if I succed with this method, but if someone has another solution, I´ll be glad to hear it.

ROBOTIS-zerom, 2019-10-29:
If the Red LED stays on, the device seems to be in bootloader mode.
I don't know how this can happen. 
Did you succeed firmware recovery?

johncgh22, 2019-10-29:
Hi @ROBOTIS-zerom 
I tried Firmware Recovey, but no success at all. I used  Dynamixel Wizard 2.0 and Wizard 1.0 (the one inside RoboPlus). I dont know if this can be fixed using a previous version of the firmware, and then upgrading again, but I don´t know if theres a repository for old firmwares.

ROBOTIS-Will, 2020-09-17:
This issue thread will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-09-17 by ROBOTIS-Will