ROBOTIS-GIT/DynamixelSDK #347: [3.7.21] Cannot read and write at all in Linux

mipsan, 2019-09-26:
1. SDK: 3.7.21
 - OS: Ubuntu 18.04 
 - HW: NVIDIA Jetson Nano 
2. Motor: MX-12W
3. converter: USB2Dynamixel2
4. I cannot get any thing from python test example:
 DynamixelSDK/python/tests/protocol1_0/read_write.py
----
> Succeeded to open the port
> Succeeded to change the baudrate
> [TxRxResult] There is no status packet!
> Press any key to continue! (or press ESC to quit!)
> [TxRxResult] There is no status packet!
> [TxRxResult] There is no status packet!
> [ID:001] GoalPos:010  PresPos:000
> Press any key to continue! (or press ESC to quit!)
> [TxRxResult] There is no status packet!
> [TxRxResult] There is no status packet!
> [ID:001] GoalPos:4000  PresPos:000
> [TxRxResult] There is no status packet!
> [ID:001] GoalPos:4000  PresPos:000

...
----

I tested this at windows 10 with Wizard SW.

This makes NVIDIA jetson Nano based robot, kaya not work with following errors:

> 2019-09-26 11:26:11.470 ERROR   packages/dynamixel/gems/dynamixel.cpp@33: Dynamixel error (-3001): [RxPacketError] Input voltage error!
> 

Any kinds of help would be appreciated.

Thanks

johncgh22, 2019-10-03:
Hi @mipsan 

It seems that the motor has a problem with voltage He is not **receving enough input voltage**. According to the specs, it recommends **12 V for optimal operation**, you can chek it here:
[http://emanual.robotis.com/docs/en/dxl/mx/mx-12w/](url)

Check how are you powering the motor. In my case I´m using a SMPS2Dynamixel Adapter [https://www.trossenrobotics.com/store/p/5886-SMPS2Dynamixel-Adapter.aspx](url) 

Also, I recommend cheking some basics in Dynamixel Wizard 2.0:

1. Check ID and baudrate.
2. Verify if the motor has the latest firmware version.
3. Verify if you can change Goal Position inside Dynamixel Wizard 2.0

After that, remember to change the settings in the read_write.py example with the values you set in Dynamixel Wizard 2.0 (ID, Baudrate, Port and Goal Positions) at the begining of the example. Use the Control Table and verify the correct **Address** 

Check this link from the e-Manual, it may help you:
[http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/python_read_write_protocol_1_0/#python-protocol-10](url)

mipsan, 2019-10-07:
Hi @johncgh22,

I could clear this issue by modifying some values for control tables.
And power feeding is OK, 12.x V. 
But I cannot control the motos From jetson Nano yet. 
I have to check some other parts to resolve it.

Thanks,

closed 2019-10-07 by mipsan