ROBOTIS-GIT/DynamixelSDK #348: [3.7.21][MATLAB2017b][USB2Dynamixel][AX-18A] Issue: failed to loadlibrary

asheesh-2019, 2019-09-26:
Facing issue in readwrite.m sample program.
error loading library.

Checked multiple times, initialized the library and all setup properly and verified.

error message
"Error using loadlibrary
There was an error loading the library "dxl_x64_c"
The specified module could not be found.

Error in test3_motor (line 48)
    [notfound, warnings] = loadlibrary(lib_name, 'dynamixel_sdk.h', 'addheader', 'port_handler.h',
    'addheader', 'packet_handler.h');

Caused by:
    Error using loaddefinedlibrary
    The specified module could not be found."

what I believe the issue to be is in the zip file DYNAMIXELSDK 3.7.21. 
as per dynamixel emanual, while library setup for matlab on windows ,  the matlab path must be setup to include corresponsding library through the folder " [DynamixelSDK folder]/c/include" which should have :
1 folder ([DynamixelSDK folder]/c/include) and 
3 subfoders ([DynamixelSDK folder]/c/include/dynamixel_sdk, [DynamixelSDKfolder]/c/include/dynamixel_sdk_linux and [DynamixelSDK folder]/c/include/dynamixel_sdk_windows )

However, the zip file I have downloaded contains:
only one folder ([DynamixelSDK folder]/c/include) and 
1 subfoder ([DynamixelSDK folder]/c/include/dynamixel_sdk)

I have followed other procedure correctly. and getting the error.

Please help me out. 

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1], [3.5.2], etc.

- I'm using `C++` Language

  - [C] / [C++] / [C#] / [Python] / [Java] / [MATLAB] / [LABVIEW] / [ROS] / [Arduino]

- I'm using `USB2Dynamixel2` serial converter

  - [USB2Dynamixel] / [USB2Dynamixel2] or [OpenCM904] / [OpenCR] / [CM700s] / [CM500s] or other ROBOTIS product only

- I'm using `XM430-W210-R`

  - [MX-28], [AX-12W], [H54-200W], etc.

- and I'm having an `issue` like what

  - 'My motor doesn't work', etc.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

johncgh22, 2019-10-03:
Hi @asheesh-2019 

This error happens because MATLAB uses **x64 stuff.** If you follow the exact steps on the e-Manual for the Dynamixel SDK, it only install **the x32 version** of the SDK. 

To resolve it, you need to add this folder into **Set Path inside MATLAB**.
[image]

With that folder added, now you can run the examples in MATLAB.

asheesh-2019, 2019-10-07:
hi @johncgh22 

it worked.

Thanks..

closed 2019-10-07 by asheesh-2019