ROBOTIS-GIT/DynamixelSDK #352: Problem with reading multiple values with GroupBulkRead (protocol 1).

zakharov, 2019-10-10:
ISSUE TEMPLATE ver. 1.1.1

Hi.

- I'm using latest `DynamixelSDK` for  `C++`  and Protocol 1.0

- My hardware setup: `USB2Dynamixel2` serial converter, [MX-64, MX-28, MX-106]

I have a problem with C++ SDK GroupBulkRead for Protocol 1.0. It works only when requesting the same parameter from multiple dynamixels (for example current position), but does not return the values when requesting multiple different parameters (for example current position and current load). When requesting multiple parameters getData  always returns the value for the parameter with the largest address and zeros for everything else. For example, request to read the current position (address: 36) and current load (address: 40) would return only values for address 40.

ROBOTIS-zerom, 2019-10-10:
Hello, zakharov.
If an identical ID is designated multiple times, only the first designated parameter will be processed.
example) If you want to read address 36 and 40 of ID 1, you have to read 6 bytes from address 36 of ID 1. (36, 37, 38, 39, 40, 41 -> 36, 37 : current position, 40, 41 : current load)
If you try to read 2 bytes from address 36 of ID 1, and read 2 bytes from address 40 of ID 1, that's the wrong way. The same ID must appear only once in one bulkread packet.

zakharov, 2019-10-11:
Indeed I have been using an identical ID for reading multiple parameters. I will try what you have suggested and let you know.

Thanks!

closed 2020-01-21 by ROBOTIS-zerom