ROBOTIS-GIT/DynamixelSDK #354: [3.7.11][C][USB2Dynamixel2][mx-64] Issue : groupSyncRead getdata failed in Protocol 2

ryans799, 2019-10-24:
- I'm using `DynamixelSDK ver. 3.7.11`
- I'm using `C` Language
- I'm using `USB2Dynamixel2` serial converter
- I'm using `MX-64
- and I'm having an `issue` : groupSyncRead getdata failed while I tried example sync_read_write "$~/DynamixelSDK-3.7.11/c/example/protocol2.0/sync_read_write/linux64" in Protocol 2.0.
[image]

ryans799, 2019-10-25:
[image]
that is my parameter, i changed the parameter address from dynamixel PRO (default example protocol 2) into parameter mx 64 in protocol 2.

ryans799, 2019-10-25:
Actually I tried sync_read_write in Protocol 1, it's success. but when I tried sync_read_write in protocol 2, it failed and shown "groupSyncRead getdata failed".

I have 4 dynamixels mx-64.
I chekced sync_read_write with 2 dynamixels MX64(firmware version=39) for protocol 1.
I chekced sync_read_write with 2 dynamixels MX64(firmware version=43) for protocol 2.

ROBOTIS-zerom, 2019-10-29:
Hi, ryans
In MX-64(2.0), the Goal Position and the Present Position are 2 bytes.
Modify the length values for lines 47, 48 from 4 to 2.

ryans799, 2019-10-29:
I have modified the length values for lines 47, 48 from 4 to 2
[image]

but the motor didn't work well. the instruction couldn't write the motor 
[image]

ROBOTIS-zerom, 2019-10-29:
I'm sorry. I made a mistake looking at the MX-64 Control Table.
The Goal Position and Present Position of the MX-64(2.0) are 4 bytes. 

I have no problem with it.

[image]

[image]

ryans799, 2019-11-05:
can I ask the source code which you used? I'll try it with my dynamixel mx-64(2.0)

ROBOTIS-zerom, 2019-11-05:
The part I changed in the master branch is the capture image above.
Only six lines. 

See the attachment here:
[sync_read_write.zip](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/3807763/sync_read_write.zip)

ryans799, 2019-11-05:
where is "#define LEN_PRO_PRESENT_POSITION" in your code?
in my code, there is "#define LEN_PRO_PRESENT_POSITION" in line 48.
is "#define LEN_PRO_PRESENT_POSITION" omitted?

ROBOTIS-zerom, 2019-11-05:
In the attachment, the code is in line 48.

ryans799, 2019-11-05:
I have tried again with your code. but the result is just the same as me. "groupSyncRead getdata failed".
[image]
sometimes it's success, and sometimes it's failed. I don't know why it happens. and I don't have any solution to repair this bug.

ROBOTIS-Will, 2020-09-17:
This issue thread will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-09-17 by ROBOTIS-Will