ROBOTIS-GIT/DynamixelSDK #358: How to sending data with the minus value?

kuochiyoug, 2019-11-03:
Hi, I recently using Dynamixel Motor for my personal project. 
I am trying to do with current control.

I notice that the data can be used in syncWrite or BulkWrite method only can accept the argument of unsigned int type that cannot use minus value at all.
However, the R+ Manager or Dynamixel Wizard 2.0 can actually test with the minus value. 

Can anyone tell me how to do with it?

kuochiyoug, 2019-11-03:
Additionally detail. I now using C# as my coding base.

ROBOTIS-zerom, 2020-01-07:
Hi, kuochiyoug.
If you want to convert unsigned int to signed int, try explicit casting.
```C#
UInt16 x = 65521;
var y = (Int16)x;    // y = -15
```

If you want to convert signed int to unsigned int, there are few ways..
```C#
Int16 x = -15;
var y1 = Convert.ToUInt16(x);
var y2 = checked((UInt16) x);
var y3 = unchecked((UInt16) x);
var y4 = (UInt16) x;
```

ROBOTIS-zerom, 2020-01-21:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2020-01-21 by ROBOTIS-zerom

amahrou, 2020-03-28:
That's not correct. If you test your answer you'll see that you simply get an exception (System.OverflowException: 'Value was either too large or too small for a UInt16.')!
the sdk api requires a uint16. How do you use write2ByteTxRx and pass a negative value when the parameter is a uint16.