ROBOTIS-GIT/DynamixelSDK #359: [3.7.21 (master branch)][C++][CM-530 converter][MX-12W] Issue : My motor doesn't work`

ipa-mah, 2019-11-04:
ISSUE TEMPLATE ver. 1.1.1

- I'm using `DynamixelSDK master branch `

  - Maybe [3.7.21]

- I'm using `C++` Language

- I'm using `CM-530`  converter

- I'm using `MX-12W` 
- and I'm getting anything from  `/DynamixelSDK/c++/example/protocol2.0/read_write/linux64/.read_write`

Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] Incorrect status packet!
Press any key to continue! (or press ESC to quit!)
[TxRxResult] There is no status packet!
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:-150000  PresPos:000
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:-150000  PresPos:000
[TxRxResult] There is no status packet!
[ID:001] GoalPos:-150000  PresPos:000
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:-150000  PresPos:000
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:-150000  PresPos:000
[TxRxResult] There is no status packet!

- I test CM-530 converter with MX-12W by using RoboPlus on Windows and it works, but it does not work on Linux using Dynamixel SDK.
Do you have any idea about the problem?
Thanks!

ipa-mah, 2019-11-18:
pingggggggggggggg

ROBOTIS-zerom, 2020-01-06:
Hi, ipa-mah.
Sorry for late answer.
What is CM-530 converter? 
Did you just connect it like  [PC] -(USB)- [CM530] -(3pin cable)- [MX12W] and test it in R+ Manger?
CM-530 is not serial converter. R+ Manager does not use Dynamixel Protocol packets.
To control using Dynamixel SDK, serial converter such as USB2Dynamixel or U2D2 should be used.

ipa-mah, 2020-01-07:
Hi @ROBOTIS-zerom ,
I'm having holidays. I will try your solution when I'm back to work and let you know the result

closed 2020-01-31 by ROBOTIS-zerom