ROBOTIS-GIT/DynamixelSDK #364: Why there are no conflicts on the serial line when all the motor report the position simultaneously?

itfanr, 2019-11-29:
Hello everyone,

I made a 6-dof robot with dynamixel motors. I found that they communication with eath other thougth serial.

The developer can send a broadcast command to get all the positions of them, which called sync group read.

What confused me is that why we did not encounter conflicts when all the motor report their position simultaneously?

Thank you.

https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/302

http://www.robotis.us/dynamixel-xm430-w350-r/

ROBOTIS-zerom, 2020-01-06:
Hi, itfanr.
When using sync read, each device respnds in the order of the ID conatained in the instruction packet. Therefore, if one device does not respond in the middle, all IDs do not respond after that.

itfanr, 2020-01-06:
@ROBOTIS-zerom  Thank you for your reply. But as the instruction packet is a broadcast command, how does the motors know the list of ID or know eath other?

ROBOTIS-zerom, 2020-01-07:
Because parameter interpretation is different depending on the type of instruction.

http://emanual.robotis.com/docs/en/dxl/protocol2/#sync-read
[image]

In the example:
[image]

P5 is the first device ID, P6 is the second device ID.

itfanr, 2020-01-11:
@ROBOTIS-zerom  Thank you. I will read the protocol carefully.

closed 2020-01-16 by itfanr