ROBOTIS-GIT/DynamixelSDK #366: [3.7.21][MATLAB] Issue: Error using load library

chjwong, 2019-12-15:
I am trying to control 8 Dynamixel motors via MATLAB on macOS Mojave.
I tried running the sample code read_write.m [DynamixelSDK folder]/matlab/protocol2.0,
but it gave the following error message

> Error using loadlibrary
> There was an error loading the library "libdxl_mac_c"
> dlopen(libdxl_mac_c.dylib, 6): image not found

> Error in read_write (line 48)
>    [notfound, warnings] = loadlibrary(lib_name, 'dynamixel_sdk.h', 'addheader',
>    'port_handler.h', 'addheader', 'packet_handler.h');

> Caused by:
>    Error using loaddefinedlibrary
>    dlopen(libdxl_mac_c.dylib, 6): image not found

I have also tried making the Makefile in [DynamixelSDK folder]/c/build/Mac/ to build the dynamic library, and the following error message from terminal appeared:

> make
> mkdir -p ./.objects/
> gcc -O2 -O3 -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -fPIC -g -c ../../src/dynamixel_sdk/group_bulk_read.c -o .objects/group_bulk_read.o
> gcc -O2 -O3 -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -fPIC -g -c ../../src/dynamixel_sdk/group_bulk_write.c -o .objects/group_bulk_write.o
> gcc -O2 -O3 -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -fPIC -g -c ../../src/dynamixel_sdk/group_sync_read.c -o .objects/group_sync_read.o
> gcc -O2 -O3 -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -fPIC -g -c ../../src/dynamixel_sdk/group_sync_write.c -o .objects/group_sync_write.o
> gcc -O2 -O3 -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -fPIC -g -c ../../src/dynamixel_sdk/packet_handler.c -o .objects/packet_handler.o
> gcc -O2 -O3 -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -fPIC -g -c ../../src/dynamixel_sdk/port_handler.c -o .objects/port_handler.o
> gcc -O2 -O3 -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -fPIC -g -c ../../src/dynamixel_sdk/protocol1_packet_handler.c -o .objects/protocol1_packet_handler.o
> gcc -O2 -O3 -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -fPIC -g -c ../../src/dynamixel_sdk/protocol2_packet_handler.c -o .objects/protocol2_packet_handler.o
> In file included from ../../src/dynamixel_sdk/protocol2_packet_handler.c:24:
> In file included from ../../include/dynamixel_sdk/protocol2_packet_handler.h:23:
> ../../include/dynamixel_sdk/packet_handler.h:104:25: error: conflicting types
>       for 'reboot'
> WINDECLSPEC void        reboot              (int port_num, int protocol_...
>                         ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/unistd.h:702:6: note: 
>       previous declaration is here
> int      reboot(int);
>          ^
> 1 error generated.
> make: *** [.objects/protocol2_packet_handler.o] Error 1

How should libdxl_mac_c.dylib be properly created?

ROBOTIS-zerom, 2020-01-07:
There was an error due to a duplicate function name with MacOSX SDK.
The issue was fixed. Please check again.

closed 2020-01-07 by ROBOTIS-zerom

jalpanchal, 2021-04-19:
I am facing the same issue from Dynamixel 3.7.31

CPS-zhangX, 2021-06-12:
> I am facing the same issue from Dynamixel 3.7.31 and in 3.7.51.
> 
> UPDATE : It did not run by default but I was able to make the library from DynamixelSDK-3.7.51/c/build
> using $make and was able to use it after that.
> 
> Thank you.

Have you solved this problem?

CPS-zhangX, 2021-06-12:
> > I am facing the same issue from Dynamixel 3.7.31 and in 3.7.51.
> > UPDATE : It did not run by default but I was able to make the library from DynamixelSDK-3.7.51/c/build
> > using $make and was able to use it after that.
> > Thank you.
> 
> Have you solved this problem?

I am facing the same problem....... Thanks!

ROBOTIS-Will, 2021-06-14:
@CPS-zhangX 
Have you built the library?
https://youtu.be/zGhd727NkJY

If you are running MATLAB on Windows : https://youtu.be/1VWksgPqyDs

jalpanchal, 2021-06-14:
@CPS-zhangX Yes, like @ROBOTIS-Will  pointed out and I mentioned in the answer update, I was able to start using it be building the library in mac and ensuring all the correct paths are included.

addpath(genpath('./../../DynamixelSDK-3.7.51/matlab/'))
addpath(genpath('./../../DynamixelSDK-3.7.51/c/build/mac/'))
addpath(genpath('./../../DynamixelSDK-3.7.51/c/include/'))

PR #368 (2020-01-07): - fixed "conflicting types for 'reboot'" error in mac OS.
fixed #366