ROBOTIS-GIT/DynamixelSDK #373: [3.5.4][Arduino][OpenCM904+485EXP][XM540-W270-R] Writing acceleration, velocity and position in one packet

ercknz, 2020-01-17:
I am working a a force controlled robot and have not been able to find a way to write accelerations, velocities, and positions. Currently I have my robot running in position control and it works by sync writing positions but i would like to control more about each movement. Is there a different mode or function I should be using? Is there a way to at least write position and velocity in one packet? Or is it position OR velocity depending on the mode only.

ROBOTIS-zerom, 2020-01-31:
We recommended that you read the manual first.
http://emanual.robotis.com/docs/en/dxl/x/xm540-w270/

XM540 Operating Mode : http://emanual.robotis.com/docs/en/dxl/x/xm540-w270/#operating-mode

If you want to write the acceleration, velocity and position values in one packet, you can also use sync write instruction.
http://emanual.robotis.com/docs/en/dxl/protocol2/#sync-write

The profile acceleration address is 108 and the length is 4 bytes.
The profile velocity address is 112 and the length is 4 bytes.
The goal position address is 116 and the length is 4 bytes.

So, you can write a 12 bytes from address 108 using syncwrite instruction.
[image]
starting address : 108
data length : 12

ercknz, 2020-01-31:
Oh I see. I think my issue was that I was feeding velocities to the velocity profile without changing it from time-based profile. Thank you.

closed 2020-01-31 by ercknz