ROBOTIS-GIT/DynamixelSDK #374: [3.7.21][MATLAB][USB2Dynamixel][MX-28] Issue :write negative values for position in multiturn mode

mmp1993, 2020-01-20:
Hey,  I have four dynamixel MX-28 that are connected together. I need to get them negative positions in multiTurn mode in Matlab but when giving them negative goal position get below error:

Error using calllib
Value exceeds the legal range for the data type.

I use protocol 1.0

ROBOTIS-zerom, 2020-01-31:
Perhaps you used the write4ByteTxRx() function to write the goal position value.
```C
void write4ByteTxRx(int port_num, int protocol_version, uint8_t id, uint16_t address, uint32_t data)
```
You must pass in an uint32_t value as the data value.

You can change the int32_value to uint32_value as in the code below:
```MATLAB
uint32_value = typecast(int32_value, 'uint32');
```

mmp1993, 2020-02-01:
Thanks for the answer
I followed your instructions as you say
but The number obtained for goal position is out of range of the daynamixel position range

[image]

mmp1993, 2020-02-01:
I use below code
[one_motor_multiTurn.zip](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/4142272/one_motor_multiTurn.zip)

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
clc;
clear;
%% Libraries
lib_name = '';

if strcmp(computer, 'PCWIN')
  lib_name = 'dxl_x86_c';
elseif strcmp(computer, 'PCWIN64')
  lib_name = 'dxl_x64_c';
elseif strcmp(computer, 'GLNX86')
  lib_name = 'libdxl_x86_c';
elseif strcmp(computer, 'GLNXA64')
  lib_name = 'libdxl_x64_c';
elseif strcmp(computer, 'MACI64')
  lib_name = 'libdxl_mac_c';
end

% Load Libraries
if ~libisloaded(lib_name)
    [notfound, warnings] = loadlibrary(lib_name, 'dynamixel_sdk.h', 'addheader', 'port_handler.h', 'addheader', 'packet_handler.h');
end
%%
% Control table address
ADDR_MX_TORQUE_ENABLE       = 24;          
ADDR_MX_GOAL_POSITION       = uint16(30);
ADDR_MX_PRESENT_POSITION    = 36;

% Protocol version
PROTOCOL_VERSION            = 1.0;         

% Default setting

DXL_ID                      =  uint8(1)   ;    
dxl_goal_position           = int32(-5000) ;
dxl_goal_position = typecast(dxl_goal_position, 'uint32');

BAUDRATE                    = 1000000;
DEVICENAME                  = 'COM3';   

TORQUE_ENABLE               = 1;            % Value for enabling the torque
TORQUE_DISABLE              = 0;            % Value for disabling the torque
DXL_MINIMUM_POSITION_VALUE  = 4095;          % Dynamixel will rotate between this value
DXL_MAXIMUM_POSITION_VALUE  = 4095;         % and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
DXL_MOVING_STATUS_THRESHOLD = 50;           % Dynamixel moving status threshold

COMM_SUCCESS                = 0;            % Communication Success result value
COMM_TX_FAIL                = -1001;        % Communication Tx Failed
%% Initialize PortHandler Structs
% Set the port path
% Get methods and members of PortHandlerLinux or PortHandlerWindows
port_num = portHandler(DEVICENAME);

% Initialize PacketHandler Structs
packetHandler();

%% Open port
openPort(port_num);

%% Set port baudrate
setBaudRate(port_num, BAUDRATE);

%% Enable Dynamixel Torque
write2ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_ENABLE);

    while 1
    %% Write goal position
    write4ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_MX_GOAL_POSITION, dxl_goal_position);

    %% Read present position
    dxl_present_position = read2ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_MX_PRESENT_POSITION);
    fprintf('[ID:%03d] GoalPos:%03d  PresPos:%03d\n', DXL_ID, dxl_goal_position, dxl_present_position);

            if ~(abs(dxl_goal_position - dxl_present_position) > DXL_MOVING_STATUS_THRESHOLD)
                break;
            end
    end
    
% end

% %% Disable Dynamixel Torque
%  write1ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_DISABLE);
%% Close port
closePort(port_num);
% Unload Library
unloadlibrary(lib_name);

% close;
% clear;

ROBOTIS-zerom, 2020-02-17:
The value read must be changed from uint32_value to int32_value.
unsigned int32 : 4294962296  
->  signed int32 : -5000

ROBOTIS-zerom, 2020-02-17:
See the source code of the example. 

https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/matlab/protocol2.0/read_write.m

    % Write goal position
    write4ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_PRO_GOAL_POSITION, typecast(int32(dxl_goal_position(index)), 'uint32'));

...

        % Read present position
        dxl_present_position = read4ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_PRO_PRESENT_POSITION);

...

        fprintf('[ID:%03d] GoalPos:%03d  PresPos:%03d\n', DXL_ID, dxl_goal_position(index), typecast(uint32(dxl_present_position), 'int32'));

ROBOTIS-Will, 2020-09-17:
This issue thread will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-09-17 by ROBOTIS-Will