ROBOTIS-GIT/DynamixelSDK #375: Dynamixel X series Indirect address mapping

ksmola, 2020-01-22:
I'm using 7 H54P-200-S500-R Dynamixel Pro, Firmware 11 as well as 4 Dynamixel XH540-V150-R Firmware 44 on one bus. 
C application on Linux, native RS485 port.

To streamline communications I'm using Indirect address mapping to get all Dynamixels in the same address space. 
This works fine with the PRO+ servos, but the X series loses the indirect address configuration upon reboot since it's stored in RAM instead of EEPROM.

Can this be changed so we can use indirect addresses just like on the PRO+ series?

ROBOTIS-zerom, 2020-01-31:
Unfortunately, there is no way to change the indirect address mapping value from RAM to EEPROM area in the X-series. I'm sorry.

ksmola, 2020-02-01:
Thanks, I solved it by (re)setting all register configurations after startup.

closed 2020-02-01 by ksmola