ROBOTIS-GIT/DynamixelSDK #377: [3.6.0][Python3][U2D2][XM430-W210-R & AX12] Issue : My motor eventually stops`

sysark-yohan, 2020-02-06:
I have the following problem which appears randomly. And, I fail to understand that it is the reason:

'''
Exception in thread Thread-3:
Traceback (most recent call last):
  File "C:\Users\Yohan\AppData\Local\Programs\Python\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Users\Yohan\Documents\Gitlab\l.e\metronic_project\metronic_app\dynamixel.py", line 310, in run
    self.setMotor()
  File "C:\Users\Yohan\Documents\Gitlab\l.e\metronic_project\metronic_app\dynamixel.py", line 605, in setMotor
    self.moveMotor(self.dictDynamixel['id_motor'].get('rot'), self.rotationMaximumPosition - pos)
  File "C:\Users\Yohan\Documents\Gitlab\l.e\metronic_project\metronic_app\dynamixel.py", line 468, in moveMotor
    dxl_present_position, dxl_comm_result, dxl_error = self.packetHandlerProtocol2.read4ByteTxRx(self.portHandler, id_motor, self.dictDynamixel['control_table_xm'].get('ADDR_XM_PRESENT_POSITION'))
  File "C:\Users\Yohan\AppData\Local\Programs\Python\Python37\lib\site-packages\dynamixel_sdk-3.6.0-py3.7.egg\dynamixel_sdk\protocol2_packet_handler.py", line 625, in read4ByteTxRx
    data[2], data[3])) if (result == COMM_SUCCESS) else 0
IndexError: list index out of range
'''

thank you

ROBOTIS-zerom, 2020-02-20:
Hi, yohan.
Sorry, but we cannot debug including your code. 
Please let me know if there is anything wrong with our example.

ROBOTIS-Will, 2020-09-17:
This issue thread will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-09-17 by ROBOTIS-Will