ROBOTIS-GIT/DynamixelSDK #378: `[3.6.0][C++/Arduino][OpenCR][DYNAMIXEL PRO+ SERIES] Issue : Cannot syncRead from Dynamixel`

StravopodisNikos, 2020-02-10:
- I'm using `DynamixelSDK ver. 3.6.0`

- I'm using `C++/ARDUINO` Language

- I'm using `OpenCR` Board as the controller. Program is uploaded using `Arduino IDE`.

- I'm using `DYNAMIXEL PRO+ SERIES` (PH54-100-S500-R and PH54-200-S500-R)

- and I'm having the following `issue` :

  - I can successfully syncWrite Items of the Control Table BUT syncRead FAILS
  - When I run any Open CR example that includes syncRead or bulkRead I get:
1.  [TxRxResult] There is no status packet!
2.  [TxRxResult] Incorrect status packet! 
  - The 2 Motors successfully reach goal Position(although I cannot see the Present Position value, I tested it mechanically).

- Details:

1. I have checked Dynamixel ID's and Baudrate(for Serial and Dynamixels is set to 2000000).
2. In debugging i see that in syncRead .addParam is successful but .txRxPacket fails.
3. I have also checked Indirect Addressing. I can also successfully Write but NOT Read.
4. I have run the code and applied different USB_LATENCY_TIMER, also set my linux USB through command:  setserial -v /dev/ttyACM0 low_latency (and for ACM1)
5. When I used ONLY one Dynamixel Motor everything worked fine.
6. I have followed the OpenCR Install on Linux for Arduino IDE as described in e-Manual
7. I have the 2 motors wired as described in the e-Manual and from the first one the signal cable is connected to one of the RS485 ports of the Board. 
8. Should I use any other library besides DynamixelSDK.h ? In DynamixelSDK eManual you mention that was tested for OpenCR. Am I missing something?

Sorry, BUT I am completely lost here. Please help.

KyoungHwan2046, 2020-02-21:
Hi, @StravopodisNikos 

I tested it with example(OpenCR/07.DynamixelSDK/protocol2.0/sync_read_write) and it works fine.

I tested on both windows and linux with PH54-100-s500-R and PH54-200-S500-R, openCR
and it does works well.

KyoungHwan2046, 2020-02-21:
@StravopodisNikos 

If you have a U2D2, then please check that two motor works with Dynamixel Wizard.

If you don't have a U2D2, then you can use the example (OpenCR/10.Etc/usb_to_dxl).

ROBOTIS-Will, 2020-09-17:
This issue thread will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-09-17 by ROBOTIS-Will