ROBOTIS-GIT/DynamixelSDK #379: [3.7.21][C++][USB2Dynamixel][XM540-W150-R] Issue: Torque Control with XM540-W150 R Dynamixel

Sabotm, 2020-02-11:
Hi everyone,
I hope someone can help me with this.

INFORMATIONS:

- I'm using 'DynamixelSDK ver. 3.1.21'
- I'm using C++ Language
- I'm using 'USB2Dynamixel' : http://www.robotis.fr/interface/283-u2d2.html
- I'm using 'XM540-W150-R'

PROBLEM:

I would like to know how to control my dynamixel by using a torque control. I succedded in using the "readwrite.cpp" code which change the position by using the goal position and the present position. But now i want to have a more fluent way by using the torque control. Has anyone already tried it? It will help me a lot!
Thank you for your answer!

ROBOTIS-zerom, 2020-02-17:
Hi, Sabotm.

If you want to control the torque, change the operating mode.
http://emanual.robotis.com/docs/en/dxl/x/xm540-w150/#operating-mode
You can use the "Current Control Mode" or "Current-based Position Control Mode".
In current control mode, you can only control current(torque) regardless of velocity and position.

Below figure is a block diagram describing the current-based position controller.
[image]

NOTE. In current control mode, Goal Current means the desired current. Therefore, the present current will be kept close to the goal current.
But, in current-based position control mode, Goal Current means the limit of the current. Therefore, the present current may be lower than the goal current.

Sabotm, 2020-02-17:
Hi! 
Thank you for your answer! I changed the operating mode from 3 to 0. I checked if the operating mode has been correctly changed and it's done. The problem is when i want to run my program to control the current, the present current doesn't change and is stuck at the value "2". 
Here is the code that I use:

`#include <stdlib.h>
#include <stdio.h>

#include "dynamixel_sdk.h"                                  // Uses Dynamixel SDK library

// Control table address
#define ADDR_MX_TORQUE_ENABLE           64                  // Control table address is different in Dynamixel model
#define ADDR_MX_GOAL_POSITION           116
#define ADDR_MX_PRESENT_POSITION        132
#define ADDR_MX_GOAL_CURRENT            102
#define ADDR_MX_PRESENT_CURRENT         126
#define ADDR_MX_OPERATING_MODE          11

// Protocol version
#define PROTOCOL_VERSION                2.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL_ID                          1                   // Dynamixel ID: 1
#define BAUDRATE                        57600 
#define DEVICENAME                      "COM6"             

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      0                 // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      4095                
#define DXL_MOVING_STATUS_THRESHOLD     10                  // Dynamixel moving status threshold
#define DXL_CURRENT_LIMIT               2047

#define ESC_ASCII_VALUE                 0x1b

int main()
{
    // Initialize PortHandler instance
    // Set the port path
    // Get methods and members of PortHandlerLinux or PortHandlerWindows
    dynamixel::PortHandler* portHandler = dynamixel::PortHandler::getPortHandler(DEVICENAME);

    // Initialize PacketHandler instance
    // Set the protocol version
    // Get methods and members of Protocol1PacketHandler or Protocol2PacketHandler
    dynamixel::PacketHandler* packetHandler = dynamixel::PacketHandler::getPacketHandler(PROTOCOL_VERSION);

    int index = 0;
    int dxl_comm_result = COMM_TX_FAIL;             // Communication result
    int dxl_operating_mode_result = COMM_TX_FAIL;             // Communication result
    int dxl_goal_current[2] = { -30, 30 };         // Goal current

    uint8_t dxl_error = 0;                          // Dynamixel error
    uint16_t dxl_present_current = 0;              // Present position
    uint8_t dxl_op_mode = 0;
    uint16_t current_limit = 0;

    // Open port
    if (portHandler->openPort())
    {
        printf("Succeeded to open the port!\n");
    }
    else
    {
        printf("Failed to open the port!\n");
        printf("Press any key to terminate...\n");
        getch();
        return 0;
    }

    // Set port baudrate
[... 88 lines cut ...]
            index = 0;
        }
    }

    // Disable Dynamixel Torque
    dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_DISABLE, &dxl_error);

    if (dxl_comm_result != COMM_SUCCESS)
    {
        printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
    }
    else if (dxl_error != 0)
    {
        printf("%s\n", packetHandler->getRxPacketError(dxl_error));
    }

    // Close port
    portHandler->closePort();

    return 0;
}`

Also what do you recommend me to choose between "Current Control Mode" or "Current-based Position Control Mode"? 

Thank you for your answer!

ROBOTIS-zerom, 2020-02-20:
You are now changing the Operating Mode after turning on the torque.
And you didn't check the dxl_comm_result after reading the Operating Mode value.

So, maybe the operating mode would not have changed.

1. turn off the torque
2. change the operating mode
3. turn on the torque
4. write the goal current

"Current Control Mode" and "Current-based Position Control Mode" operate completely differently. 
Position control is not available in the "Current Control Mode".
In the "Current-based Position Control Mode", it is only a position control mode in which the current value does not exceed a certain level, and cannot maintain a constant current(torque) value.
Therefore, you should choose according to your operating intention.

Sabotm, 2020-02-20:
Hi! Thank you for your answer! I change the operating mode before the enabling torque and it worked so thank you!
[Current Control Mode] For the goal current, i tried with the value "40", the present value is around "20" it moves  and stop when i try to block the rotation and when the current is over 40 which is normal. So it acts more like a limit rather than an actual goal but in the datasheet, it's said that the goal current is used to choose a desired current/torque. How is it possible? 

Thank you for the time that you are taking to answer to my questions,
Have a nice day

shoaib2527, 2020-04-09:
@Sabotm 
Hello, I'm doing something really similar. 
Can you share updated code?

ROBOTIS-Will, 2020-09-17:
This issue thread will be closed as there isn't recent activity. Please feel free to reopen when necessary.

closed 2020-09-17 by ROBOTIS-Will