ROBOTIS-GIT/DynamixelSDK #380: [3.3.3][Python][USB2DYNAMIXEL2][MX-64] Issue: Dynamixels lose calibration on restart

shantnavagarwal, 2020-02-17:
Details: 

3 Dynamixel MX 64 connected to Raspberry Pi via U2D2. All running Protocol 1.0 in multi-turn mode.

I have set the offset of the motor (via Dynamixel Wizard) such that the position of the motor remains between values 0 - 3000. However on restart, this calibration breaks. The offset value does not get reset but writing a value between 0-3000 causes the motor to go to a wrong position (out of the bounds allowable by my robot’s structure).

The present position as reported by the SDK function is very high (~65000) just after motor restart instead of a value between (0-3000) as expected. This probably causes the motor to rotate multiple rounds. 

How can i fix this?

ROBOTIS-zerom, 2020-02-17:
For example, 
If you set the multi turn offset to -2000, and resolution divider to 1, 

- present position = (real position / 1) - 2000
- real position = present position + 2000

If the present position is 3000, the real position is 5000. (in multi turn mode, 1 lap(4096) + 904)
When you reset the motor, the information about 1 lap is lost. So the real position value is 904.

- present position = 904(real position) - 2000(multi turn offset) = -1096

Therefore, the present position value, which was 3000 before reset, is changed to -1096 after reset.
That is, if the multi turn offset value is large and the real position value is out of the range of 0 ~ 4095, the present position value after reset will be different from the value before reset.

shantnavagarwal, 2020-02-17:
Yes, this makes sense. I have gotten an idea from your response on how to fix my issue. Thanks

I am closing the issue now but shall reply with results.

closed 2020-02-17 by shantnavagarwal

shantnavagarwal, 2020-02-18:
I ended up using the motors in the joint mode only!

grndctrl2majtom, 2020-11-01:
@ROBOTIS-zerom I don't follow exactly, can the resolution divider be used to save calibration data after a full rotation?

ROBOTIS-zerom, 2020-11-04:
@grndctrl2majtom I don't understand your question. 

See the link below for a description of the resolution divider.
https://emanual.robotis.com/docs/en/dxl/mx/mx-64/#resolution-divider

hannibal-paul, 2022-02-11:
I know its late but this solution worked for me in Python for MX-28. Useful when you want to automatically set the current position to 0.
1. Set,  Multi Turn Offset = 0
2. Read, Present Position
3. Set, Multi Turn Offset = -Present Position

Any position you send after this will be with respect to the current position.