ROBOTIS-GIT/DynamixelSDK #381: `[3.7.21][Python][USB2Dynamixel2][XL430-W250-T] Issue: Motor Rotation and float bytearray allocation`

tahir1069, 2020-02-25:
- I'm using `DynamixelSDK ver. 3.7.21`

- I'm using `Python` Language

- I'm using `OpenCR` serial converter

- I'm using `XL430-W250-T`

I have a differential drive robot with 4 active wheels (left_motors ->[1, 3], right_motors -> [2, 4]). I am programming in Python two control motors in velocity control mode. I have following questions:

1. My motor # 4 rotate opposite to motor # 1. I don't know why as this is not the case with motor # 3 with respect to motor # 1. The motors are daisy chained and the motor # 4 is the last one in the chain (I have changed the motor but the issue is the same).

2. For writing to Servos a byte array is used of the form `[DXL_LOBYTE(DXL_LOWORD(dxl_goal_position)),DXL_HIBYTE(DXL_LOWORD(dxl_goal_position)),DXL_LOBYTE(DXL_HIWORD(dxl_goal_position)),DXL_HIBYTE(DXL_HIWORD(dxl_goal_position))]`. This configuration works fine for integers but it is not possible to use this for float. Can you tell me any work around which can resolve this issuse. I have used Python bytearray method as [here](https://stackoverflow.com/questions/36893206/converting-a-float-to-bytearray) but this does not resolve the issue.

3. I really can not understand the differnce between [drive mode](http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#drive-mode) and [operating mode](http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#operating-mode11). For me operating mode it is defining the control type of the motor which is obvious but what is drive mode and how it is related to operating mode and also to velocity control mode. 

    3.1. If you can point to any example of drive mode, it will really be appreciated.

tahir1069, 2020-03-11:
@ROBOTIS-OFFICIAL can anyone please reply.

Also can you elaborate how to get encoder ticks, I think it can be read from [present position](http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#present-position). But how to decode this value is what am not able to understand and stuck. Any leads in this regard as well will highly be appreciated.

Thank you so much in advance.

Hello970, 2021-05-28:
Hello, brother, has the second problem been solved? I currently meet the same problem.

ROBOTIS-Will, 2021-05-28:
@Hello970 You can only write integer to DYNAMIXEL. Could you elaborate more about your issue?

Hello970, 2021-05-28:
@ROBOTIS-Will 
Hi, sir, thanks for your prompt reply. I will give a specific description of my issue.

- I'm using DynamixelSDK ver. 3.7.31
- I'm using Python Language
- I'm using a U2D2 converter
- I'm using MX64-AR

I would like to control three Dynamixel motors with position control mode. I try to use the "sycn_read_write" method to send the goal position message to each motor.

At first, I implement it with MATLAB code. It means I can control three motors to goal positions (The goal position for each motor is different). Now, due to additional requirements, I try to use Python code for motor control. However, as a new user, I am not familiar with some grammar rules. Therefore, I met some errors after modifying MATLAB code to Python version. The main errors are as follows.

1. Type Error: object of the type 'float' has no len( );
For this error, I modified it by using str(int(goal_position)). It worked. However, I met the second problem.

2. List index out of range.
I can understand the meaning of the error, but I haven't fix it well.

For further understanding, my current code is attached. Thanks for your attention.

Attached code

#!/usr/bin/env python
# -*- coding: utf-8 -*-

################################################################################

# *********     Sync Read and Write for Three Motor Control      *********

################################################################################

import os, time

if os.name == 'nt':
    import msvcrt
    def getch():
        return msvcrt.getch().decode()
else:
    import sys, tty, termios
    fd = sys.stdin.fileno()
    old_settings = termios.tcgetattr(fd)
    def getch():
        try:
            tty.setraw(sys.stdin.fileno())
            ch = sys.stdin.read(1)
        finally:
            termios.tcsetattr(fd, termios.TCSADRAIN, old_settings)
        return ch

from dynamixel_sdk import *                     # Uses Dynamixel SDK library

def tic():
    #Homemade version of matlab tic and toc functions
    import time
    global startTime_for_tictoc
    startTime_for_tictoc = time.time()

def toc():
    import time
    if 'startTime_for_tictoc' in globals():
        print('Elapsed time is ' + str(time.time() - startTime_for_tictoc) + ' seconds.')
    else:
        print('Toc: start time not set')

# Control table address
ADDR_MX_HOMING_OFFSET           = 20            # Control table address is different in Dynamixel model
ADDR_MX_TORQUE_ENABLE           = 64
ADDR_MX_LED                     = 65
ADDR_MX_PROFILE_ACCELERATION    = 108
ADDR_MX_PROFILE_VELOCITY        = 112
[... 197 lines cut ...]
groupSyncRead.clearParam()

# Disable Dynamixel#1 Torque
dxl_comm_result, dxl_error = packetHandler.write1ByteTxRx(portHandler, DXL1_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_DISABLE)
if dxl_comm_result != COMM_SUCCESS:
    print("%s" % packetHandler.getTxRxResult(dxl_comm_result))
elif dxl_error != 0:
    print("%s" % packetHandler.getRxPacketError(dxl_error))

# Disable Dynamixel#2 Torque
dxl_comm_result, dxl_error = packetHandler.write1ByteTxRx(portHandler, DXL2_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_DISABLE)
if dxl_comm_result != COMM_SUCCESS:
    print("%s" % packetHandler.getTxRxResult(dxl_comm_result))
elif dxl_error != 0:
    print("%s" % packetHandler.getRxPacketError(dxl_error))

# Disable Dynamixel#3 Torque
dxl_comm_result, dxl_error = packetHandler.write1ByteTxRx(portHandler, DXL3_ID, ADDR_MX_TORQUE_ENABLE, TORQUE_DISABLE)
if dxl_comm_result != COMM_SUCCESS:
    print("%s" % packetHandler.getTxRxResult(dxl_comm_result))
elif dxl_error != 0:
    print("%s" % packetHandler.getRxPacketError(dxl_error))

# Close port
portHandler.closePort()

ROBOTIS-Will, 2021-05-31:
Hi @Hello970 

Unfortunately, we do not review users' custom code.
Have you tried running the default example included in DYNAMIXEL SDK?
https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol2_0/sync_read_write.py
I think first try it and modify from the example might easier for you.
- **DYNAMIXEL + Python Quick Start Guide Video**: https://youtu.be/LAizFTTdL8o

Also, when appending the parameter as below, you should use little endian.
```
dxl_addparam_result = groupSyncWrite1.addParam(DXL1_ID, str(int(dxl1_goal_position)))
```
In the above code, `str(int(dxl1_goal_position))` is using an incorrect value and you should use something like below to make it little endian
```
param_goal_position = [DXL_LOBYTE(DXL_LOWORD(dxl1_goal_position)), DXL_HIBYTE(DXL_LOWORD(dxl1_goal_position)), DXL_LOBYTE(DXL_HIWORD(dxl1_goal_position)), DXL_HIBYTE(DXL_HIWORD(dxl1_goal_position))
dxl_addparam_result = groupSyncWrite1.addParam(DXL1_ID, param_goal_position)
```

Hello970, 2021-05-31:
Dear Sir,

Thanks for your prompt reply.

Yes, I ran the demo in DYNAMIXEL SDK at first. It was OK when I synchronically sent goal position to three motors just in the given position range (for example from 150 ~ 2000). Also, the little endian has been used since the demo gives the similar forms. However, it didn’t work when I modify the goal position according to the real requirement. I gave the error “the float has no len()”. Therefore, I change it into “str(int())”. Although it eliminated the errors about data type, it cased another problem. I will consider effective solutions to solve this problem.

Thank you all the same.

Best regards,
Haonan Sun

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol2_0/sync_read_write.py

ROBOTIS-Will, 2021-05-31:
It is because you divided angle values with 360 as below.
```
ini_W_position = ini_W_angle / 360 * 4096 + offset_W # Wheel
ini_C_position = ini_C_angle / 360 * 4096 + offset_C # Carrier
ini_S_position = ini_S_angle / 360 * 4096 + offset_S # Sun gear
```
You need to convert the output into an integer then apply the little endian format to pass into the addparam() function.

Hello970, 2021-05-31:
Dear Sir,

Thanks for your patience.

According to your guidance, I have successfully solved this problem.

Thank you very much. I would be grateful for your help.

Best regards,
Haonan Sun

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol2_0/sync_read_write.py

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-13 by yun-goon