ROBOTIS-GIT/DynamixelSDK #383: [3.7.21][C++][CustomSerialAdapter][MX28/XM430-W210-T/XL320] Issue : no status packet only when sending rate is high

peclatj, 2020-03-08:
Hello,

I'm trying to write multiple values to a servomotor, maximum torque, p gain, offset, etc...
I get some "[TxRxResult] There is no status packet!" when I write two packets too close from each other in time, but this only happens to ~1/6 of the packets sent in a row and it is often, but not always, the same packet that fails when I run multiple times my program. If I put a "usleep(10000)" in the "PortHandlerLinux::writePort()" function it is slow but it works, all the packets are sent correctly.

The dynamixelSDK thinks it has sent the packet and is waiting for the status packet until the time is out, but when I check on the servomotor, the value in the RAM is still the same as before. The motor probably ignored or didn't receive the packet, and of course didn't respond.

For the context, I'm using your code on debian on a Texas Instruments microcontroller (BeagleBoneBlack), we made an adapter to put the TX and RX on a single cable. The baudrate of the servomotors is 57160. I have 12 motors on two daisy chains:

UART ->
MX28 (the packets to this one fails often) -> MX28 -> XM430 -> XM430 -> XM430 -> XM430 -> XM430
XL320 -> XL320 -> XL320 -> XL320 (the packets to this one fails often) -> XL320
Here is the schematics of our adapter, the idea is just that we don't want to hear ourself when we send data.
[image]

In other words we took a look at this image, and we made the TX itself acts as the direction port.
[image]

Have you experiment such a behaviour ? I don't think it comes from our eletronics because I have never seen a "corrupt packet" error message.

Thank you for your advices and have a nice day
Jonathan P.

ROBOTIS-Will, 2020-09-04:
@peclatj 
Hi Jonathan, I'm sorry for this super late reply, but hope this information helps you.
Almost all cases when the DYNAMIXEL SDK is applied on a custom board, TX / RX flow control causes major problems.
If you have a packet analyzer or oscillator, check if your DIRECTION_PORT signal correctly responds after TX and before RX.
Some MCU may require additional time for actually transmit the data on the wire after setting a byte on the TX buffer.
Thank you.

ROBOTIS-Will, 2021-01-29:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will