ROBOTIS-GIT/DynamixelSDK #386: `[3.7.21][MATLAB][USB2Dynamixel][AX-12A and AX-18A] Issue : Can't use the bulk read

Ugurmustafa97, 2020-03-09:
Hi,

- I'm using `DynamixelSDK ver. 3.7.21 `
- I'm using `MATLAB` Language
- I'm using `USB2Dynamixel2` serial converter
- I'm using `AX-12A and AX-18A`
- and I'm having an `issue that is I can't use the bulk read functionality` 

I am working to create a robot arm using four DYNAMIXEL servo motors. I am trying to move them to follow the trajectory required. So far, I have been moving good.

However, the system that I build works more slowly than I except. So, I think that maybe that delay results from the while loop that I read all the servo motors' positions then I tried to use the bulk read structure to read positions of them instead of reading each of their positions values separately.

However, I always get the error in the following:

```
Dynamixel1 has been successfully connected 
Dynamixel2 has been successfully connected 
Dynamixel3 has been successfully connected 
Dynamixel4 has been successfully connected 
Press any key to go the goal positions! (or input e to quit!)

Speed of the first motor has been successfully changed. 
Speed of the second motor has been successfully changed. 
Speed of the third motor has been successfully changed. 
Speed of the fourth motor has been successfully changed. 
[TxRxResult] There is no status packet!
[ID:001] groupBulkRead getdata failed
```

I also tried to run the sample code in the MATLAB samples called `bulk_read.m` and I get the same error in every time. The following is the error I get when I run it:

```
Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected 
Dynamixel has been successfully connected 
Press any key to continue! (or input e to quit!)

[TxRxResult] There is no status packet!
[ID:001] groupBulkRead getdata failed
```

Finally, I am using Protocol1.0. And, if you need it I can share the complete code file.

Thanks,
Mustafa

ROBOTIS-zerom, 2020-03-30:
Hi, Mustafa.
http://emanual.robotis.com/docs/en/dxl/protocol1/#instruction
In protocol 1.0, the BulkRead instruction can only be used with MX series.
AX-series cannot use the bulk read instruction.

Ugurmustafa97, 2020-04-02:
Okay, thank you so much for the information.

closed 2020-04-02 by Ugurmustafa97