ROBOTIS-GIT/DynamixelSDK #389: [3.7.21][C++][USB2Dynamixel][XM540-W150-R] Issue: Torque Control with XM540-W150 R Dynamixel and PWM Control Mode

Sabotm, 2020-03-29:
Hi everyone,
I hope someone can help me with this.

INFORMATIONS:

    I'm using 'DynamixelSDK ver. 3.1.21'
    I'm using C++ Language
    I'm using 'USB2Dynamixel' : http://www.robotis.fr/interface/283-u2d2.html
    I'm using 'XM540-W150-R'

PROBLEM:

I would like to know how to control my dynamixel by using a torque control. I thought that the Current Control Mode was the perfect control mode to do it but when i use it, the "present_current" value is never the same as my "goal_current" value. I understood that " goal_current" value is not a goal but a limit so that's why my motor stop when i put some forces on it and it's present_current is higher than the goal_current.
So, ROBOTIS told me to use the PWM mode to do an external loop control on the current but i tried and i got the same problem : "present_current" is never equal or close to the value that i want. Has someone ever tried to use the PWM mode to control the current? How did you do it ?

Thank you for your answer and take care !

Sabotm

ROBOTIS-Will, 2020-09-04:
@Sabotm 
Sorry, this is super late reply, but hope this information helps you to resolve some of your issues.
If you want to keep certain torque with a set current, for example, gripping an object with a gripper attached to DYNAMIXEL, you can use Current-based Position Control mode.
With this mode, you can set a certain Goal Position to close the gripper, then it will refer to Goal Current to adjust the force of gripping.

FYI, when DYNAMIXEL is set to Current Control Mode, it uses the Goal Current value as a reference to rotate the horn to certain direction. The present current may sometimes exceed this Goal Current as it senses the actual current flowing on the core motor, but exceeding this Goal Current should not stop the DYNAMIXEL unless there's a shutdown condition.

Thank you.

ROBOTIS-Will, 2021-01-29:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will