ROBOTIS-GIT/DynamixelSDK #392: [3.7.21][Python][*][*] Empty __init__.py

psh117, 2020-04-23:
- I'm using `DynamixelSDK in ROS distribution`
`ros-melodic-dynamixel-sdk` `3.7.21-1bionic.20200320.104450`

- I'm using `Python` Language
- I'm using `USB2Dynamixel2` serial converter

- I'm using `XM430-W350`

- and I'm having an `issue` like what

  - The installed python package `/opt/ros/melodic/lib/python2.7/dist-packages/dynamixel_sdk` has  empty __init__.py and it causes import error. I guess that #391 has the same problem.

[image]

ROBOTIS-zerom, 2020-05-27:
It will be applied when deploying ROS packages later. Probably after August.
So, if necessary, it is recommended to download and use the source directly from github master branch.

psh117, 2020-06-02:
Thanks! I'm using yours by modifying the `__init__.py` in the ROS distribution package directory. Everything seems to be working fine with that modification.

closed 2020-11-17 by ROBOTIS-zerom