ROBOTIS-GIT/DynamixelSDK #396: [Python] [Linux]NameError: name 'PortHandler' is not defined

lmaddira, 2020-05-23:
- I'm using `DynamixelSDK ver. 3.7.21`

- I'm using `python` Language - Linux

- I'm using `U2D2` serial converter with OpenCM 485 EXP

- I'm using AX-12A and I also tried with 2XL-430-W250

when I try to run the source code of read_write.py I get an error as below:

portHandler = PortHandler(DEVICENAME)
NameError: name 'PortHandler' is not defined

lmaddira, 2020-05-23:
looks like  msvcrt is not importing and this seems to be a particular issue in Linux 
did anyone else faced this issue??
I tried the same program in window it worked . How do I get this working in Linux

lmaddira, 2020-05-23:
Is this a particular issue with Linux. As I ran the same in Windows and it works

ROBOTIS-zerom, 2020-05-26:
I'm using LinuxMint 19.3.
I ran it according to the link below and there was no problem.

http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/python_linux/#python-linux

- DynamixelSDK master branch
- Serial Converter : U2D2
- Dynamixel : 2XC430-W250

[image]
[image]

lmaddira, 2020-05-26:
Hey I'm using Ubuntu 18.4
I did the same as you did but still the problem persists
[image]
and 
[image]
Can you think of any obvious problem?

ROBOTIS-zerom, 2020-05-27:
I tried the same test on ubuntu 18.04 using docker.  (using DynamixelSDK 3.7.21)
Still no problems.. 
[image]

I don't know what the cause is because the problem doesn't happen.

ROBOTIS-zerom, 2020-05-27:
If you comment out the 50th line of importing Dynamixel SDK library from read_write.py file, the same error occurs. Therefore, a problem that can be considered is when the Dynamixel SDK libraray is not properly imported.

lmaddira, 2020-05-27:
Yeah that can be an issue. How to rectify it?
I tried to run the same in windows and it works. But I can't use windows. How can I get this working in my system?
I tried to just get PortHandler working on the terminal as below. Still the same issue
[image]

billynugrahas, 2020-06-03:
try the sudo python sync_write.py

ShotaAk, 2020-07-24:
I had same error on Ubuntu 18.04 and dynamixel_sdk==3.7.31.
And I have quick fixed the error by editing as bellow:

```diff
# read_write.py

from dynamixel_sdk import *                    # Uses Dynamixel SDK library
+ from dynamixel_sdk.port_handler import PortHandler
+ from dynamixel_sdk.packet_handler import PacketHandler
+ from dynamixel_sdk.robotis_def import *
```

---

Installation log:

```sh
➜  python git:(master) sudo python setup.py install
running install
running bdist_egg
running egg_info
creating src/dynamixel_sdk.egg-info
writing requirements to src/dynamixel_sdk.egg-info/requires.txt
writing src/dynamixel_sdk.egg-info/PKG-INFO
writing top-level names to src/dynamixel_sdk.egg-info/top_level.txt
writing dependency_links to src/dynamixel_sdk.egg-info/dependency_links.txt
writing manifest file 'src/dynamixel_sdk.egg-info/SOURCES.txt'
reading manifest file 'src/dynamixel_sdk.egg-info/SOURCES.txt'
writing manifest file 'src/dynamixel_sdk.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/dynamixel_sdk
copying src/dynamixel_sdk/packet_handler.py -> build/lib.linux-x86_64-2.7/dynamixel_sdk
copying src/dynamixel_sdk/protocol1_packet_handler.py -> build/lib.linux-x86_64-2.7/dynamixel_sdk
copying src/dynamixel_sdk/group_bulk_read.py -> build/lib.linux-x86_64-2.7/dynamixel_sdk
copying src/dynamixel_sdk/protocol2_packet_handler.py -> build/lib.linux-x86_64-2.7/dynamixel_sdk
copying src/dynamixel_sdk/port_handler.py -> build/lib.linux-x86_64-2.7/dynamixel_sdk
copying src/dynamixel_sdk/group_sync_write.py -> build/lib.linux-x86_64-2.7/dynamixel_sdk
copying src/dynamixel_sdk/group_sync_read.py -> build/lib.linux-x86_64-2.7/dynamixel_sdk
copying src/dynamixel_sdk/robotis_def.py -> build/lib.linux-x86_64-2.7/dynamixel_sdk
copying src/dynamixel_sdk/__init__.py -> build/lib.linux-x86_64-2.7/dynamixel_sdk
copying src/dynamixel_sdk/group_bulk_write.py -> build/lib.linux-x86_64-2.7/dynamixel_sdk
creating build/bdist.linux-x86_64
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/dynamixel_sdk
copying build/lib.linux-x86_64-2.7/dynamixel_sdk/packet_handler.py -> build/bdist.linux-x86_64/egg/dynamixel_sdk
copying build/lib.linux-x86_64-2.7/dynamixel_sdk/protocol1_packet_handler.py -> build/bdist.linux-x86_64/egg/dynamixel_sdk
copying build/lib.linux-x86_64-2.7/dynamixel_sdk/group_bulk_read.py -> build/bdist.linux-x86_64/egg/dynamixel_sdk
copying build/lib.linux-x86_64-2.7/dynamixel_sdk/protocol2_packet_handler.py -> build/bdist.linux-x86_64/egg/dynamixel_sdk
copying build/lib.linux-x86_64-2.7/dynamixel_sdk/port_handler.py -> build/bdist.linux-x86_64/egg/dynamixel_sdk
copying build/lib.linux-x86_64-2.7/dynamixel_sdk/group_sync_write.py -> build/bdist.linux-x86_64/egg/dynamixel_sdk
copying build/lib.linux-x86_64-2.7/dynamixel_sdk/group_sync_read.py -> build/bdist.linux-x86_64/egg/dynamixel_sdk
copying build/lib.linux-x86_64-2.7/dynamixel_sdk/robotis_def.py -> build/bdist.linux-x86_64/egg/dynamixel_sdk
copying build/lib.linux-x86_64-2.7/dynamixel_sdk/__init__.py -> build/bdist.linux-x86_64/egg/dynamixel_sdk
copying build/lib.linux-x86_64-2.7/dynamixel_sdk/group_bulk_write.py -> build/bdist.linux-x86_64/egg/dynamixel_sdk
byte-compiling build/bdist.linux-x86_64/egg/dynamixel_sdk/packet_handler.py to packet_handler.pyc
byte-compiling build/bdist.linux-x86_64/egg/dynamixel_sdk/protocol1_packet_handler.py to protocol1_packet_handler.pyc
byte-compiling build/bdist.linux-x86_64/egg/dynamixel_sdk/group_bulk_read.py to group_bulk_read.pyc
byte-compiling build/bdist.linux-x86_64/egg/dynamixel_sdk/protocol2_packet_handler.py to protocol2_packet_handler.pyc
byte-compiling build/bdist.linux-x86_64/egg/dynamixel_sdk/port_handler.py to port_handler.pyc
byte-compiling build/bdist.linux-x86_64/egg/dynamixel_sdk/group_sync_write.py to group_sync_write.pyc
byte-compiling build/bdist.linux-x86_64/egg/dynamixel_sdk/group_sync_read.py to group_sync_read.pyc
byte-compiling build/bdist.linux-x86_64/egg/dynamixel_sdk/robotis_def.py to robotis_def.pyc
byte-compiling build/bdist.linux-x86_64/egg/dynamixel_sdk/__init__.py to __init__.pyc
byte-compiling build/bdist.linux-x86_64/egg/dynamixel_sdk/group_bulk_write.py to group_bulk_write.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying src/dynamixel_sdk.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying src/dynamixel_sdk.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying src/dynamixel_sdk.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying src/dynamixel_sdk.egg-info/requires.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying src/dynamixel_sdk.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/dynamixel_sdk-3.7.31-py2.7.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
Processing dynamixel_sdk-3.7.31-py2.7.egg
Copying dynamixel_sdk-3.7.31-py2.7.egg to /usr/local/lib/python2.7/dist-packages
Adding dynamixel-sdk 3.7.31 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/dynamixel_sdk-3.7.31-py2.7.egg
Processing dependencies for dynamixel-sdk==3.7.31
Searching for pyserial==3.4
Best match: pyserial 3.4
Adding pyserial 3.4 to easy-install.pth file

Using /usr/lib/python2.7/dist-packages
Finished processing dependencies for dynamixel-sdk==3.7.31
```

ShotaAk, 2020-07-25:
I found my mistake... The module path was incorrect. :cry: 

My python script imports the Dynamixel SDK module from ROS (*Robot Operating System*) PATH **not** manually installed path.

---
How to check the path:

```sh
➜  ~ python3
Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dynamixel_sdk
>>> dynamixel_sdk.__file__
'/opt/ros/melodic/lib/python2.7/dist-packages/dynamixel_sdk/__init__.py'
```
---

I have no idea how to fix this incorrect path (I don't want to break my environments.)

I hope my post helps you. :+1:

lmaddira, 2020-08-07:
Thanks for the post.....Looks like its the same issue with me it takes the path from ROS PATH not manually installed path. Did you figure out how to take care of it ??

ShotaAk, 2020-08-08:
> Thanks for the post.....Looks like its the same issue with me it takes the path from ROS PATH not manually installed path. Did you figure out how to take care of it ??

@lmaddira 
I found a solution! :laughing: 

## 1. Edit `.zshrc` (`.bashrc`) file like this:

```sh
echo -n "ROS or ROS2 ?(1:ROS, 2:ROS2, other:nothing):"
read USE_ROS
if [ ${USE_ROS} = "2" ]; then
    echo "Use ROS2"
    # ROS2 Settings
    export ROS_DISTRO=dashing
    source /opt/ros/$ROS_DISTRO/setup.zsh
    # hogehogehoge
elif [ ${USE_ROS} = "1" ]; then
    echo "Use ROS"
    # ROS1 Settings
    export ROS_DISTRO=melodic
    source /opt/ros/$ROS_DISTRO/setup.zsh
    # hogehogehoge
fi
```
This scripts able to change the Python PATH at login.

## 2. Install the dynamixel_sdk library for Python 3 correctly.

```sh
$ cd /the/dynamixel_sdk/python/path
$ sudo python3 setup.py install
```

Dear @ROBOTIS-zerom .
The command in official documents `sudo python setup.py install` did not work for **Python 3** in Ubuntu 18.04.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/python_linux/#python-linux

```sh
# My python command is alias of python3
➜  ~ which python
python: aliased to python3
```

## 3. Complete!

The above setup results like this:

```sh
ROS or ROS2 ?(1:ROS, 2:ROS2, other:nothing):n
➜  ~ python3
Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dynamixel_sdk
>>> dynamixel_sdk.__file__
'/usr/local/lib/python3.6/dist-packages/dynamixel_sdk-3.7.31-py3.6.egg/dynamixel_sdk/__init__.py'
>>> 
```

```sh
ROS or ROS2 ?(1:ROS, 2:ROS2, other:nothing):1 
Use ROS
➜  ~ python3
Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import dynamixel_sdk
>>> dynamixel_sdk.__file__
'/opt/ros/melodic/lib/python2.7/dist-packages/dynamixel_sdk/__init__.py'
>>> 
```

ROBOTIS-Will, 2021-01-29:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will