ROBOTIS-GIT/DynamixelSDK #399: [3.7.0][C][U2D2][XM430-W210-T, XL430-W210-T] Issue : XL430-W210-T and XM430-W210-T Commutability

KevinGim, 2020-07-03:
Hello,

I have a question about XL430-W210-T and XM430-W210-T commutability.

My system is configured as follows:

Board: Orange PI lite 2 (Armbian)
Programming language: C
Servos: 12 XM430-W210-T & 2 XL320 in a serial hub

I am trying to replace XM430 servos with XL430 servos.

I operated the system without any problem under 1ms of Latency timer which can be changed in “port_handler_linux.c” of the Dynamixel SDK.
I also reduced the “return delay time” of each dynamixels from 250ms to 5ms. With the setting, I was able to achieve 8ms loop time for Sync read/write of 12 XL430 servos.

However, when I replaced the servos with XL430 dynamixels, I found that communication getting very unstable.
The number of connected servo increases, some servos are disconnected occasionally. Especially, when I connect 12 XL430s, the system detects 12 servos by serial port scan, but two servos lost connection when it enters the main loop. 

As a solution to the problem, I increased the Latency timer to 8ms and it worked but with a longer loop time of 17ms (for 12 servos). 
Here, I have a question that why this difference happens although the XM series and XL series have the same MCU, ARM CORTEX-M3 (72 MHZ, 32 Bit).

Thank you,

ROBOTIS-Will, 2020-07-23:
@KevinGim 
Hi, could you provide any test code that we can reproduce the issue you are experiencing?
At least we'd like to know the series of functions you are using to control 12 DYNAMIXELs.
Thank you.

ROBOTIS-Will, 2021-01-29:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will