ROBOTIS-GIT/DynamixelSDK #423: Matlab support for DynamixelSDK on Mac OS

shardulamrite, 2020-07-16:
- I'm using `DynamixelSDK ver. 3.7.21
- I'm using `MATLAB 2019.A
- I'm using U2D2` serial converter
- I'm using `XM430-W350-R
-  I am trying to control the Dynamixel Motors through MATLAB on my MacBook. There are no instruction on how to build the C library for MacOS on the e - manual on the Robotis site. Could you please give the detailed instructions needed to complete the setup in order to control the Dynamixels through Matlab on a MacOS ? There are instructions provided for windows and linux but not for MacOS

ROBOTIS-Will, 2020-08-21:
Hi,
Thank you for the information. We'll continue to build up the DYNAMIXEL SDK.
Please refer to Linux instructions as the Mac instructions will be very similar to it except slight variations such as `.so` -> `.dylib` and compile and build processes.

ROBOTIS-Will, 2021-01-29:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will