ROBOTIS-GIT/DynamixelSDK #429: [3.7.3][Python3][Half-Duplex UART to TTL][AX-12A] Issue: I cannot run any of the software without seeing this error >>> "[TxRxResult] There is no status Packet!"

silver2row, 2020-07-26:
ISSUE TEMPLATE ver. 1.1.1

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.7.3`

  - [3.7.3]

- I'm using the `Python` Language

  - [Python]

- I'm using `Half-Duplex UART to TTL` serial converter

  - [Half-Duplex UART to TTL]

- I'm using `AX-12A`

  - [AX-12A]

- and I'm having an `issue` like this:

  - `elif result == COMM_RX_TIMEOUT:
            return "[TxRxResult] There is no status packet!"`
    This bit of software on the Protocol1 source for the Dynamixel SDK seems to be canceling out my RX receiver.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Hello...here: https://community.robotsource.org/t/re-half-duplex-uart-for-ttl-commincation-and-wiring-w-the-hc126-and-hc04/1844/7. I am using some equipment to handle GPIO and Half-Duplex UART to TTL for use with a AX-12A from my beagleboard.org SBC running Linux (Debian 10). I keep receiving the above error since I have not written any Python3 source.

...

I tried to run ping.py as `python3 ping.py` but to no avail. I have not been able to run any tests successfully so far.

- Does the SDK work as is or do I need to add some source? I am asking b/c I have not come across any source so far for testing the DynamixelSDK.

    from robotis_def import *
    from time import sleep

Would source for Python3 start like this for the DynamixelSDK or is there another way to initialize the source?

Seth

silver2row, 2020-07-30:
Hello,

I heard from a fellow on this forum. I will test the forum idea and relay service back to this post.

Seth

silver2row, 2020-08-02:
Hello,

I tested ideas from the forum person and I have not been able to write up a set of source to communicate to my actuator from my setup and linux_sbc. 

...

Is there a way to compile C/C++ source onboard my linux_sbc? Is there are starter script to run to test the actuator(s)?

Seth

P.S. With compiling the C/C++ source, it works with the Makefile file but when I try to compile, for instance, ping.c/cpp, it does not work. Errors are evident.

ROBOTIS-zerom, 2020-08-19:
Hi, 
First, it is necessary to check whether there is any problem in communication by hardware. 
The community seems to be discussing this issue.
If there is no problem with hardware, follow the link below to configure and run.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/python_linux/#python-linux 

I haven't tried compiling the SDK with the C/C++ compiler for ARM.
What error did you get?

GMDFr, 2020-09-03:
Hi, 
I have got the same issue using : 
Dynamixel SDK 3.7.31
Jetson Nano Jetpack 4.4 or PC Linux 
Ubuntu 18.04 (both)
linux_sdb
U2D2 with SMPS2DXL
Config 1 : AX12A or Config 2 : MX28 (not plugged in same time)
Using c++ exemple.
Compiled using g++ 7

First, when I compiled c++ from the build folder, I have got this (Jetson Nano and PC Linux) :
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_bulk_read.cpp -o .objects/group_bulk_read.o
../../src/dynamixel_sdk/group_bulk_read.cpp: In member function ‘bool dynamixel::GroupBulkRead::getError(uint8_t, uint8_t*)’:
../../src/dynamixel_sdk/group_bulk_read.cpp:235:19: warning: suggest parentheses around assignment used as truth value [-Wparentheses]
   return error[0] = error_list_[id][0];
          ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_bulk_write.cpp -o .objects/group_bulk_write.o
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_sync_read.cpp -o .objects/group_sync_read.o
../../src/dynamixel_sdk/group_sync_read.cpp: In member function ‘bool dynamixel::GroupSyncRead::getError(uint8_t, uint8_t*)’:
../../src/dynamixel_sdk/group_sync_read.cpp:204:19: warning: suggest parentheses around assignment used as truth value [-Wparentheses]
   return error[0] = error_list_[id][0];
          ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_sync_write.cpp -o .objects/group_sync_write.o
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/packet_handler.cpp -o .objects/packet_handler.o
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/port_handler.cpp -o .objects/port_handler.o
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/protocol1_packet_handler.cpp -o .objects/protocol1_packet_handler.o
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/protocol2_packet_handler.cpp -o .objects/protocol2_packet_handler.o
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/port_handler_linux.cpp -o .objects/port_handler_linux.o
g++ -shared -fPIC -m64 -o ./libdxl_x64_cpp.so ./.objects/group_bulk_read.o ./.objects/group_bulk_write.o ./.objects/group_sync_read.o ./.objects/group_sync_write.o ./.objects/packet_handler.o ./.objects/port_handler.o ./.objects/protocol1_packet_handler.o ./.objects/protocol2_packet_handler.o ./.objects/port_handler_linux.o -lrt

Anyway, it's compiled, I installed using make install.

Then, I compiled the c++ example (protocol1/dxl_monitor):
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -I../../../include/dynamixel_sdk -m64 -g -c ../dxl_monitor.cpp -o .objects/dxl_monitor.o
../dxl_monitor.cpp: In function ‘int main(int, char**)’:
../dxl_monitor.cpp:405:10: warning: ignoring return value of ‘char* fgets(char*, int, FILE*)’, declared with attribute warn_unused_result [-Wunused-result]
     fgets(input, sizeof(input), stdin);
     ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -I../../../include/dynamixel_sdk -m64 -g  .objects/dxl_monitor.o -o dxl_monitor -ldxl_x64_cpp -lrt

When I use dxl_monitor, I have got this error, nothing is detected while everything is plugged and usb permission is ok : 
 ***********************************************************************
*                            DXL Monitor                              *
***********************************************************************

Succeeded to open the port!

 - Device Name : /dev/ttyUSB0
 - Baudrate    : 57600

[CMD] scan

Scan Dynamixel Using Protocol 1.0
............................................................................................................................................................................................................................................................

Scan Dynamixel Using Protocol 2.0
............................................................................................................................................................................................................................................................

When I used read_write protocol1 or protocol2, I have this (Jetson Nano & PC linux) :
Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!
Press any key to continue! (or press ESC to quit!)
[TxRxResult] There is no status packet!
[TxRxResult] There is no status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] There is no status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] There is no status packet!
[... 67 lines cut ...]
[TxRxResult] There is no status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] There is no status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] There is no status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] There is no status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] There is no status packet!
[ID:001] GoalPos:100  PresPos:000
[TxRxResult] There is no status packet!
[ID:001] GoalPos:100  PresPos:000
[... cut ...]
[TxRxResult] There is no status packet!
[ID:001] GoalPos:100  PresPos:000

Also, I noticed that I have problems using the hexapod_ros from KevinOchs using AX12A https://github.com/KevinOchs/hexapod_ros, servo are very jerky and lagging with a lot of warning "empty" and some line indicated the position. It finished with a stack overflow on terminal.
I wondered if I could debugged this using the exemple of DynamixelSDK.
^^ A bugg can hide an other bugg ^^

Thanks for your reply ! 
Best regards 

Guillaume

ROBOTIS-Will, 2020-09-04:
@GMDFr 
Hi, I tried with below settings and didn't have problem running it. 

- Intel NUC
- Ubuntu 18.04 
- MX-28
- U2D2 with U2D2 PHB(replaces SMPS2DXL)
- built and installed from DynamixelSDK/c++/build/linux64 library

If you haven't added your ID to dialout group for accessing USB port, please perform below command and try again.
```bash
$ sudo usermod -aG dialout <your_account_id>
```

**kim@kim-NUC8i5BEK:~/DynamixelSDK/c++/build/linux64$ make**
```bash
mkdir -p ./.objects/
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_bulk_read.cpp -o .objects/group_bulk_read.o
../../src/dynamixel_sdk/group_bulk_read.cpp: In member function ‘bool dynamixel::GroupBulkRead::getError(uint8_t, uint8_t*)’:
../../src/dynamixel_sdk/group_bulk_read.cpp:235:19: warning: suggest parentheses around assignment used as truth value [-Wparentheses]
   return error[0] = error_list_[id][0];
          ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_bulk_write.cpp -o .objects/group_bulk_write.o
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_sync_read.cpp -o .objects/group_sync_read.o
../../src/dynamixel_sdk/group_sync_read.cpp: In member function ‘bool dynamixel::GroupSyncRead::getError(uint8_t, uint8_t*)’:
../../src/dynamixel_sdk/group_sync_read.cpp:204:19: warning: suggest parentheses around assignment used as truth value [-Wparentheses]
   return error[0] = error_list_[id][0];
          ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/group_sync_write.cpp -o .objects/group_sync_write.o
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/packet_handler.cpp -o .objects/packet_handler.o
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/port_handler.cpp -o .objects/port_handler.o
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/protocol1_packet_handler.cpp -o .objects/protocol1_packet_handler.o
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/protocol2_packet_handler.cpp -o .objects/protocol2_packet_handler.o
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -c -I../../include/dynamixel_sdk -m64 -fPIC -g -c ../../src/dynamixel_sdk/port_handler_linux.cpp -o .objects/port_handler_linux.o
g++ -shared -fPIC -m64 -o ./libdxl_x64_cpp.so ./.objects/group_bulk_read.o ./.objects/group_bulk_write.o ./.objects/group_sync_read.o ./.objects/group_sync_write.o ./.objects/packet_handler.o ./.objects/port_handler.o ./.objects/protocol1_packet_handler.o ./.objects/protocol2_packet_handler.o ./.objects/port_handler_linux.o -lrt
```

**kim@kim-NUC8i5BEK:~/DynamixelSDK/c++/build/linux64$ sudo make install**
```bash
mkdir -p ./.objects/
g++ -shared -fPIC -m64 -o ./libdxl_x64_cpp.so ./.objects/group_bulk_read.o ./.objects/group_bulk_write.o ./.objects/group_sync_read.o ./.objects/group_sync_write.o ./.objects/packet_handler.o ./.objects/port_handler.o ./.objects/protocol1_packet_handler.o ./.objects/protocol2_packet_handler.o ./.objects/port_handler_linux.o -lrt
cp "./libdxl_x64_cpp.so" "/usr/local/lib/libdxl_x64_cpp.so"
ln -s "/usr/local/lib/libdxl_x64_cpp.so" "/usr/local/lib/libdxl_x64_cpp.so.2"
ln -s "/usr/local/lib/libdxl_x64_cpp.so" "/usr/local/lib/libdxl_x64_cpp.so.2.0"
ln -s "/usr/local/lib/libdxl_x64_cpp.so" "/usr/local/lib/libdxl_x64_cpp.so.2.0.0"
cp -r ../../include/dynamixel_sdk/* /usr/local/include/dynamixel_sdk
ldconfig
```

- built and executed DynamixelSDK/c++/example/dxl_monitor

**kim@kim-NUC8i5BEK:~/DynamixelSDK/c++/example/dxl_monitor/linux64$ make**

```bash
mkdir -p .objects/
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -I../../../include/dynamixel_sdk -m64 -g -c ../dxl_monitor.cpp -o .objects/dxl_monitor.o
../dxl_monitor.cpp: In function ‘int main(int, char**)’:
../dxl_monitor.cpp:405:10: warning: ignoring return value of ‘char* fgets(char*, int, FILE*)’, declared with attribute warn_unused_result [-Wunused-result]
     fgets(input, sizeof(input), stdin);
     ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -I../../../include/dynamixel_sdk -m64 -g  .objects/dxl_monitor.o -o dxl_monitor -ldxl_x64_cpp -lrt
```

**kim@kim-NUC8i5BEK:~/DynamixelSDK/c++/example/dxl_monitor/linux64$ ./dxl_monitor **
```bash
***********************************************************************
*                            DXL Monitor                              *
***********************************************************************

Succeeded to open the port!
[... 29 lines cut ...]
 
 ==================== Commands for Dynamixel Protocol 2.0 ====================
 
 wrb2|w2 [ID] [ADDR] [VALUE] :Write byte [VALUE] to [ADDR] of [ID]
 wrw2 [ID] [ADDR] [VALUE]    :Write word [VALUE] to [ADDR] of [ID]
 wrd2 [ID] [ADDR] [VALUE]    :Write dword [VALUE] to [ADDR] of [ID]
 rdb2 [ID] [ADDR]            :Read byte value from [ADDR] of [ID]
 rdw2 [ID] [ADDR]            :Read word value from [ADDR] of [ID]
 rdd2 [ID] [ADDR]            :Read dword value from [ADDR] of [ID]
 r2 [ID] [ADDR] [LENGTH]     :Dumps the control table of [ID]
[... cut ...]
...........................................................................................................................................................................................................................................................

Scan Dynamixel Using Protocol 2.0
............................................................................................................................................................................................................................................................
```

GMDFr, 2020-09-04:
@ROBOTIS-Will 
Hi, 

First, thank's you for your reply.

I tried different things but I can't understand why it doesn't work with the specific baudrate 57600.
I set up the baudrate to 1000000 and it worked.

experimental@experimental-MS-7B46:~/DynamixelSDK$ sudo adduser experimental dialout
[sudo] password for experimental: 
The user `experimental' is already a member of `dialout'.

experimental@experimental-MS-7B46:~/DynamixelSDK$ ls -l /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 sept.  4 09:14 /dev/ttyUSB0

experimental@experimental-MS-7B46:~/DynamixelSDK$ id -Gn
experimental adm dialout cdrom sudo dip plugdev lpadmin sambashare

experimental@experimental-MS-7B46:~/DynamixelSDK$ sudo chmod a+rw /dev/ttyUSB0
[sudo] password for experimental:

experimental@experimental-MS-7B46:~/DynamixelSDK$ ls -l /dev/ttyUSB0
crw-rw-rw- 1 root dialout 188, 0 sept.  4 09:14 /dev/ttyUSB0

***********************************************************************
*                            DXL Monitor                              *
***********************************************************************

Succeeded to open the port!

 - Device Name : /dev/ttyUSB0
 - Baudrate    : 57600

[CMD] scan

Scan Dynamixel Using Protocol 1.0
............................................................................................................................................................................................................................................................

Scan Dynamixel Using Protocol 2.0
............................................................................................................................................................................................................................................................

[CMD] baud 1000000
 Success to change baudrate! [ BAUD RATE: 1000000 ]
[CMD] scan

Scan Dynamixel Using Protocol 1.0

 [ID:001] Model No : 00029                ... SUCCESS 
...........................................................................................................................................................................................................................................................

Scan Dynamixel Using Protocol 2.0
............................................................................................................................................................................................................................................................

[CMD]

ROBOTIS-Will, 2020-09-07:
@GMDFr 
Thank you for sharing more details.
Unlike the X series, the AX series use 1,000,000bps as [a default baudrate](https://emanual.robotis.com/docs/en/dxl/ax/ax-12a/#baud-rate-4).
Therefore, you should set your port baudrate to 1,000,000 in order to properly detect and communicate with AX series.
Thank you.

silver2row, 2020-09-18:
Hello,

Default baudrate is 1,000,000 bps. Okay. I think I put it as 9600 and tried to turn it on via a GPIO pin for my UART to TTL communication. 

...

I am almost set up again. I will test soon. 

Seth

P.S. I am using the AX-12A also as mentioned. I will go from here to the forum and back to keep others informed when using their Dynamixel Servos w/ embedded Linux machines, i.e. 32-bit SBCs.

ROBOTIS-Will, 2020-09-22:
As this issue is not directly relevant to DYNAMIXEL SDK, I'm going to close it.
Please continue the discussion in the forum below.
https://community.robotsource.org/t/re-half-duplex-uart-for-ttl-commincation-and-wiring-w-the-hc126-and-hc04/1844
Thank you.

closed 2020-09-22 by ROBOTIS-Will