ROBOTIS-GIT/DynamixelSDK #430: [3.7.31][C][U2D2][XH540-W270-R] Issue : Baudrate 4.5M does not work

robotiscustomer, 2020-08-05:
ISSUE TEMPLATE ver. 1.1.1

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.7.31`

- I'm using `C` Language

- I'm using `U2D2` serial converter

- I'm using `XH540W270-R`

- and I'm having an `issue`

I am trying to modify the sample program in DynamixelSDK/c/example/protocol2.0/read_write to suit my project.

I set the Baudrate of a XH540-W270R via the Linux version of DynamixelWizard 2.0 to 4.5M. After modifying the include parameters for XH540-W270R, I set the Baudrate to 4500000. However, when I run the program, I got:

Succeeded to open the port!
[PortHandlerLinuxLLsetCustomBaudrate] Cannot set speed to 4500000, closest is 4800000
Failed to change the baudrate!
Press any key to terminate...

Both U2D2 and XH540-W270 eManual mentioned that 4.5M rather than 4.8M is supported. I have also set the latency of the USB0 port to 1 ms.

STEP 2. Write `Title` as `[3.7.31][C][U2D2][XH540-W270-R] Issue : Baudrate 4.5M does not work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

After changing the Baudrate and the included parameters to match those for XH540-W270-R, I got the following error when I executed: ./read_write under  ~/Desktop/DynamixelSDK/c/example/protocol2.0/read_write/linux64

Succeeded to open the port!
[PortHandlerLinuxLLsetCustomBaudrate] Cannot set speed to 4500000, closest is 4800000
Failed to change the baudrate!
Press any key to terminate...

Here is the part in the read_write.c program that I changed:

// Control table address
#define ADDR_PRO_TORQUE_ENABLE          64                  // Control table address is different in Dynamixel model
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132

// Protocol version
#define PROTOCOL_VERSION                2.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL_ID                          1                   // Dynamixel ID: 1
#define BAUDRATE                        4500000
#define DEVICENAME                      "/dev/ttyUSB0"      // Check which port is being used on your controller
                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      0                   // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      4095                // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD     10                  // Dynamixel moving status threshold

#define ESC_ASCII_VALUE                 0x1b

ROBOTIS-Will, 2020-08-12:
Hi,
Regarding the 4.5Mbps, there was an issue with not getting a closer baudrate.
The hotfix has been applied on the `develop` branch, so please use the develop branch code or download the develop branch zip file from the link below.
https://github.com/ROBOTIS-GIT/DynamixelSDK/archive/develop.zip
Thank you.

robotiscustomer, 2020-08-14:
Hi Will, thank you. In the U2D2 eManual, under Baudrate, it states that at 4.5M, the actual BPS is 4571,428.571 with an error of 1.56 but in the XH540-W270R eManual, it states that 4.5M is supported. So, does this fix reduce the error % when using the XH540-W270R with U2D2 under Linux?

ROBOTIS-Will, 2020-08-18:
Hi,
The baudrate of each device is dependent on the hardware(MCU) they use.
Since the margin of error for 4.5Mbps between DYNAMIXEL and U2D2 is less than 3%, there's no problem running DYNAMIXEL with U2D2 at 4.5Mbps.
Thank you.

robotiscustomer, 2020-08-18:
Thanks. In the worst case, what could happen? Will the DYNAMIXEL gone crazy, move unexpectedly and damage themselves due to moving out of the expected range of motion and collide with other body parts?

ROBOTIS-zerom, 2020-08-19:
If baudrate error exceeds 3%, communication fails. 
It's like trying to communicate with devices set with different baudrate.

robotiscustomer, 2020-08-20:
Thanks. 

Since the margin is less than 3%, am I correct that I don't need to worry about this will be happening?

ROBOTIS-zerom, 2020-08-21:
https://www.ftdichip.com/Support/Knowledgebase/index.html?an232b_05calc.htm

The exact Baud rate may not be achievable - however as long as the actual Baud rate used is within +/-3% of the required Baud rate then the link should function without errors.

closed 2020-08-24 by ROBOTIS-zerom