ROBOTIS-GIT/DynamixelSDK #433: [3.7.31][C][U2D2][XH540-W270-R] Issue : Need clarifications on C Sync Write Protocol 2

robotiscustomer, 2020-08-24:
ISSUE TEMPLATE ver. 1.1.1

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE WON'T FIX ANYWAY

STEP 1. Check what you are suffering from :

I'm using DynamixelSDK ver. 3.7.31

I'm using C Language

I'm using U2D2 serial converter

I'm using XH540W270-R

and I'm having an issue

Need clarifications on C-Protocol 2.0-C Sync Write

STEP 2. Write Title as [3.7.31][C][U2D2][XH540-W270-R] Issue : Need clarifications on C Sync Write Protocol 2

STEP 3. Delete all written here, and describe what your problem is

Hi, I need some help in understanding the SDK code and API. Could you please help?

I read through the code and explanations in: /DynamixelSDK/c/example/protocol2.0/sync_read_write/sync_read_write.c and the "In C Sync Write Protocol 2.0" part of the SDK eManual:

https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/c_read_write_protocol_1_0/

Q1: In 6.1.8 C GroupSyncWrite's groupSyncWriteAddParam, shouldn't it be "Adds parameter storage for write" rather than "for read"?

Q2: groupSyncReadAddParam's description mentioned about the "Dynamixel ID list". What is it? Is it a linked list which stores the ID number of the dynamixels with the ordering first in first out? At any point in time, when assigning target position to each dynamixel and getting present position from each dynamixel, is it the list used help the functions to keep track of which target and present positions correspond to which dynamixels?

Q3: In C Sync Write Protocol 2.0 of the SDK eManual, it states: "write1ByteTxRx() function orders to the #DXL1_ID and #DXL2_ID DYNAMIXEL..." What is mean by "orders to the..."? Please clarify.

Q4: When do we need to execute groupSyncWriteClearParam() and groupSyncReadClearParam()? Do we always have to do that? In the program, groupSyncWriteClearParam() is used but groupSyncReadClearParam() is not used in the program even it is written in the program description.

Q5: In https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/api_reference/c/c_groupsyncread/#c-groupsyncread, it states that:

groupSyncReadGetData returns false when there is no data from the target address. Under what situations will there be no data from the target address? In the case of getting the present position, doesn't each encoder continuing keeps getting the readings for the program?

Q6: In the program explanation, it states that "groupSyncReadIsAvailable() function checks if available data is in the groupsyncread data storage.". What is groupsyncread data storage? Is there a pictorial illustration?

Q7: Supposing that I have X number of motors in a robot, do I have to call groupSyncReadIsAvailable X number of times to check if such groupsyncread data all motors are available and only when they are all available, I execute the function groupSyncReadGetData()? 

Q8: In the program explanation, it states that "groupSyncReadTxRxPacket() function orders to the Dynamixel #DXL1_ID and #DXL2_ID at the same time through #port_num port," What is mean by "orders to the Dynamixel..."?

Q9: What does groupSyncReadTxRxPacket() do? In the program description, it states what I wrote in Q8 but in the comment within the program, it states that it "sync read present position". Then, in the API description, it says that it "transmits and receives the package by using TxPacket and RxPacket." So what does it actually do?

Q10: In 6.1.8.1, it mentions groupData struct and group_num parameter storage. What are they? Are there pictorial illustrations? 

Thank you

ROBOTIS-Will, 2020-08-31:
Hi,

Thank you for your inquiries and please see below reponses.

1. Thank you for pointing out the typo. It'll be fixed soon.

2. The groupSyncreadAddParam simply helps to assemble the Protocol 2.0 packet, and the DYNAMIXEL ID list represents the listed ID in the [Protocol 2.0 SyncRead packet](https://emanual.robotis.com/docs/en/dxl/protocol2/#parameters-3).

3. You can replace the "orders to" with "sends an instruction to". I'll request to paraphrase the sentence.

4. The `groupSyncReadClearParam()` is implemented at the end of the `groupSyncRead()` and automatically resets the parameters and struct when initializing the groupSyncRead. You may use it before reading data using SyncRead.

5. The groupSyncReadGetData will return false when tries to read from invalid ID or address range that cannot be read from.

6. The storage mean the `GropuData` struct where the data is saved at and initialized before the syncRead.
```
typedef struct
{
  int         port_num;
  int         protocol_version;
  int         data_list_length;
  uint8_t     last_result;
  uint8_t     is_param_changed;
  uint16_t    start_address;
  uint16_t    data_length;
  DataList   *data_list;
}GroupData;
```

7. You'll simply use `groupSyncReadGetData` in order to parse the returned data with groupSyncReadIsAvailable(). During this parsing process, if there's any unavailable/invalid data, it'll return false. In the `sync_read_write.c` example, we run this groupSyncReadGetData a couple times to get the Present Position of ID1 and ID2 DYNAMIXEL.

8. You can replace the "orders to" with "sends an instruction to".

9. All instructions are consisted of request and response part, and in order to read data from DYNAMIXEL, we first need to send the request(TX) instruction then receive the response(RX) packet.

10. GroupData struct can be found in the [group_sync_write.c file](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/c/src/dynamixel_sdk/group_sync_write.c#L37). group_num is a counter that adjusts the size of the struct.
```
typedef struct
{
  int         port_num;
  int         protocol_version;
  int         data_list_length;
  uint8_t     is_param_changed;
  uint16_t    start_address;
  uint16_t    data_length;
  DataList   *data_list;
}GroupData;
```

robotiscustomer, 2020-08-31:
Hi, thank you very much for your clarifications.

In regard to:

A4: 
a)"You may use it before reading data using SyncRead." Do you mean GroupSyncRead? 

b) I am a bit confused about the order of calling groupSyncReadClearParam(), groupSyncRead() and groupSyncWriteTxPacket(groupwrite_num). What is the correct order? From your reply (if there is no typo), it seems to be:
groupSyncRead->groupSyncReadClearParam()-> SyncRead

c) What about the required order of execution of groupSyncWriteClearParam() and other related functions?

A7:
a) What do you mean by "parsing" the return data?
b) If I have 30 or 40 dynamixels, what is the recommended number of times to run groupSyncReadGetData? How do I determine it myself?

A10:
What does group_num corresponds to? What does it count?

ROBOTIS-Will, 2020-09-01:
Hi,

**A2**: Yes, the function [prepares various data field for the packet](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/c/src/dynamixel_sdk/group_sync_write.c#L141) to be written to DYNAMIXEL.

**A4**:
a) right, I meant groupSyncRead.

b) After gropuSyncWriteTxpacket, you should explicitly perform groupSyncWriteClearParam or else the next groupSyncWriteAddParam may be appended to the previous packet.

c) I'm sorry about the confusion. The **groupSyncReadClearParam** resets all data field in the struct so this should not be performed unless you want to update the groupSyncRead function to read different data from DYNAMIXEL. The correct order of using groupSyncRead based on the sync_read_write example is: groupSyncRead (initialization) -> groupSyncReadAddParam -> groupSyncReadTxRxPacket -> groupSyncReadIsAvailable -> groupSyncReadGetData.
Once groupSyncRead packet is prepared after the `groupSyncReadAddParam`, you may keep using the same packet over and over again in the loop with `groupSyncreadTxRxPacket` to read the identical memory address of DYNAMIXEL.

**A7**:
a) groupSyncReadGetData will look into the received packet and calculate the corresponding data and return it as a result. What I meant with "parsing" is getting a meaningful data from the packet.

b) SyncRead is much faster than multiple of Read, therefore, if possible, it is recommended to use SyncRead for the group of DYNAMIXEL you are going to operate at a time.

**A10**:
If you take a look at the [eManual](https://emanual.robotis.com/docs/en/dxl/protocol2/#instruction), there's no groupSyncRead or groupSyncWrite, but SyncRead and SyncWrite. In the DYNAMIXEL SDK, groupSyncRead or groupSyncWrite feature based on SyncRead and SyncWrite allows to easily control multi-joint robots. For example, if the left arm DYNAMIXEL is defined by groupSyncWrite with group_num = 0, you can write to the DYNAMIXEL on the left arm with the assigned group_num = 0. This helps to reduce the unnecessary data flow for unused joints.

Thank you.

robotiscustomer, 2020-09-01:
Thank you very much for your clarifications. This is getting interesting.

A4C: So, an example for needing to perform groupSyncReadClearParam more than once is to get the rotational angle of all DYNAMIXEL in the robot. Then, perform groupSyncReadClearParam before getting the torque of all DYNAMIXEL in the robot. In the next time step, perform groupSyncReadClearParam to get the rotation angle of all DYNAMIXEL, etc.?

A10: 
i)So supposing that there are 4 DYNAMIXEL in the left arm, 4 DYNAMIXEL in the right arm, 6 DYNAMIXEL in the left leg, 6 DYNAMIXEL in the right leg and 2 DYNAMIXEL in the neck. We can then use group_num = 0 to lump all four DYNAMIXEL in the left arm as one group, group_num = 1 to lump all 4 DYNAMIXEL in the right arm as the 2nd group, group_num = 2 and group_num = 3 to respectively lump all 5 DYNAMIXEL in the left and right leg. Then, use group_num = 4 to lump the two DYNAMIXEL in the neck? Given that all the joints in the robot is moving all the time, under what situation do we use this method, rather than just group_num to lump all 22 DYNAMIXEL as one group,  to help reduce the unnecessary data flow for unused joints?

ii) Do the groupread_num and groupwrite_num in the sync_read_write.c sample program  work in the same way but in the program, for example, they treat the two DYNAMIXEL as the two joints in the neck and use group_num to lump them together?

A7 OK I will use SyncRead. Thank you for the suggestion. 

i) In your first reply, you mentioned that "we run this groupSyncReadGetData a couple times to get the Present Position of ID1 and ID2 DYNAMIXEL." Based on A10 above, it looks like in the sync_read_write.c program, DXL1_ID and DXL2_ID are lumped into one group under groupread_num for reading and groupwrite_num for writing. However, in each iteration within the do-while loop, groupSyncReadIsAvailable and groupSyncReadGetData are called twice, once for DXL1_ID and once for DXL2ID. Supposing that I have 50 DYNAMIXEL in my robot, within each iteration of the do-while loop, do I have to call groupSyncReadIsAvailable and groupSyncReadGetData 50 times once for each DYNAMIXEL if I want to get data from all 50 motors? 

ii) Similarly, in the while loop of the sync_read_write.c program, write1ByteTxRx (to enable torque), groupSyncWriteAddParam are called twice once for DXL1_ID and once for DX2_ID. If I have, say 50 DYNAMIXELS, do I have to call them 50 times and then just call groupSyncWriteTxPacket only once if they are all lumped as a group under groupwrite_num?

ROBOTIS-Will, 2020-09-02:
Hi,

A4C : You only need to groupSyncReadClearParam when there're changes in ID or data field to read. Or, you can separate two different groupSyncRead with groupread_num.

A10, i) Some robots do not need to move all the joint at the same time. If you need to control all joint at once, you should use only one group_num.

ii) Not sure if I understand your question, but, the group_num is intended to identify a specific syncRead or syncWrite packet for chosen DYNAMIXEL. For example, if you include ID 1 and ID 2 Present Position for the neck with groupread_num = 4, groupSyncReadTxRxPacket(4) will only send and receive packets from ID1 and 2.

A7. i) Yes, that is what I called "parsing" data from the packet. At the moment when running the groupSyncReadGetData, the packet from selected DYNAMIXEL is already saved in the struct and we only collect necessary information from the packet. 

ii) Yes, or you can simply write a single write with [Broadcast ID of 254(0xFE)](https://emanual.robotis.com/docs/en/dxl/protocol2/#packet-id) to turn on or off all connected DYNAMIXEL.

Thank you.

robotiscustomer, 2020-09-03:
Hi, thank you for the clarification.

A4C: By "data field", what are you talking about? Do you have some examples? 
Also, under what situation will there be a change in ID? Please provide examples.

A7ii: In the "robot with 50 DYNAMIXEL example" mentioned in A7i, can we use Boardcase  ID of 254 to avoid calling groupSyncReadIsAvailable and groupSyncReadGetData 50 times (in each iteration within the do-while loop) once for each DYNAMIXEL if I want to get data from all 50 motors in the robot? It seems that we cannot do that since the ID of each DYNAMIXEL within the robot is required as the 2nd parameter of these two functions.

Thank you.

ROBOTIS-Will, 2020-09-03:
Hi,

A4C : What I meant for the data field in the param is DYNAMIXEL Addresses or Length or ID. When you want to read from different address or different ID, you can clear the read param and add new param(address, length, id) in it.

A7ii : Broadcast ID cannot be used with SyncRead or SyncWrite function as DYNAMIXEL won't return status packet for the Broadcast ID. 

Thank you.

robotiscustomer, 2020-09-04:
Hi Will,

Thank you for your clarifications. I will try the program using 2 and more DYNAMIXELS.

By the way, groupSyncWriteClearParam() is called after groupSyncWriteTxPacket() before the do-while loop. What does groupSyncWriteClearParam() do? According to the eManual's API section, it seems to clear some kind of parameter storage. Is it the write package? Do we always have to call this function after groupSyncWriteTxPacket()?

Thank you.

ROBOTIS-Will, 2020-09-04:
Hi Chun,

The groupSyncWriteClearParam() clears parameter in the SyncWrite instruction packet. If you take a look at the Protocol 2.0 [Sync Write example](https://emanual.robotis.com/docs/en/dxl/protocol2/#example-8), the packet transmits Goal Position of 150 and 170 to ID 1 and 2 respectively. In order to send new Goal Positions, we need to clear the old parameters so that the instruction packet can store updated Goal Position. This is when you need the groupSyncWriteClearParam().

Thank you.

robotiscustomer, 2020-09-07:
Hi, thank you for your clarifications. Now it became much easier to understand the sync_read_write and bulk_read_write programs.

I noticed that the the X-series has a Secondary (Shadow) ID. What are the pros and cons of using this Secondary ID vs. the method to lump different groups of DYNAMIXEL (e.g. group_num1 for all DYNAMIXEL in the left arm and group_num2 for all DYNAMIXEL in the right arm) mentioned in A10?

Shouldn't the ID(7) in Steps. 4 and 7 in Section 2.4.8.1 in the eManual for XH540-W270 be ID(12)?

If I want to include a clock in my program/robot, should I use the timer in the Linux system or whatever function your company offers? What is the Realtime Tick(12) of the DYNAMIXEL's time?

Thank you.

ROBOTIS-Will, 2020-09-07:
Hi,

Glad to hear that you find these information is helpful.
The Secondary ID has some limitations as explained in the [eManual](https://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#secondaryshadow-id12). You can only send an identical data to all DYNAMIXEL sharing the same Secondary ID.

In the [example](https://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#secondary-id12-example), the `ID(7)` in step 2, 4, 7 indicate the ID byte of instruction packet while step 1, 6 indicates the Secondary ID(12)

It is recommended to use your own system clock for the control frequency since we do not have any information about your application. In the same vein, we do not provide a specific guidance for programming custom applications. You can contact at **contactus2@robotis.com** and ask for a consultation service if needed.
The Realtime Tick is an internal clock in DYNAMIXEL that starts from 0 when powered and counts up every milliseconds and when it reaches to 32767.

Thank you.

robotiscustomer, 2020-09-09:
Thank you. 

I have been having a problem with groupSyncWriteAddParam() for the past two days. In the sample program, it is:

```
   // Add Dynamixel#1 goal position value to the Syncwrite storage
 dxl_addparam_result = groupSyncWriteAddParam(groupwrite_num, DXL1_ID, dxl_goal_position[index], LEN_PRO_GOAL_POSITION);
    if (dxl_addparam_result != True)
    {
      fprintf(stderr, "[ID:%03d] groupSyncWrite addparam failed", DXL1_ID);
      return 0;
    }

    // Add Dynamixel#2 goal position value to the Syncwrite parameter storage
    dxl_addparam_result = groupSyncWriteAddParam(groupwrite_num, DXL2_ID, dxl_goal_position[index], LEN_PRO_GOAL_POSITION);
    if (dxl_addparam_result != True)
    {
      fprintf(stderr, "[ID:%03d] groupSyncWrite addparam failed", DXL2_ID);
      return 0;
    }

```

In my program, I combined them into one using a for loop to loop through each DYNAMIXEL ID:

    for (motor_index = 1; motor_index <= NUM_DXL_MOTOR; motor_index++)
    {
      dxl_addparam_result = groupSyncWriteAddParam(groupwrite_num, DXL_ID_ARRAY[motor_index],
      target_angle[motor_index], LEN_PRO_GOAL_POSITION);
//    if (dxl_addparam_result != True)
//    {
//      fprintf(stderr, "[ID:%03d] groupSyncWrite addparam failed", DXL_ID_ARRAY[motor_index]);
//      return 0;
//    }
    }

My program runs fine for this part if I don't do any checking like above but whenever I uncomment the four lines for checking dxl_addparam_result, even the NUM_DXL_MOTOR remains to be 2, the for loop is executed more than it should be. For example, if NUM_DXL_MOTOR is 2 or 3, the for loop runs up to motor_index = 7. When NUM_DXL_MOTOR is set to 4 or 5, the for loop runs up to mator_index = 9.  Do you know what is wrong? DL_ID_ARRAY[] stores the ID of the two DYMAMIXEL I am using. 

Also, at the very end of my program, I close the port using "closePort(port_num);" and then "return 0;" just like the sample program. However, I got the following errors:

*** stack smashing detected ***: terminated
Aborted (core dumped)

Know what is wrong and how to fix it?

ROBOTIS-Will, 2020-09-11:
Hi Chun,
I'm sorry but I have no idea why your for loop iterates more than it should be.
I'd recommend to debug your code and track where the motor_index variable gets updated.
Thank you.

robotiscustomer, 2020-09-11:
Hello, this bug caused all sorts of strange behavior in my program. It is hard to debug. Perhaps due to possible misunderstanding on the use of the "return 0" statement. In the original sync_read_write.c program, could you please clarify what the "return 0" inside the if statement does?

```
    dxl_addparam_result = groupSyncWriteAddParam(groupwrite_num, DXL2_ID, dxl_goal_position[index], LEN_PRO_GOAL_POSITION);     

    if (dxl_addparam_result != True)
    {
      fprintf(stderr, "[ID:%03d] groupSyncWrite addparam failed", DXL2_ID);
      return 0;
    }
```

The bug disappeared after I commented out the "return 0;" line in my code. My program works fine when this line is gone.

Also, during the debugging, I added some printf statements in: ~/Desktop/DynamixelSDK/c/src/dynamixel_sdk/group_sync_write.c

Then, I went to ~/Desktop/DynamixelSDK/c/build/linux64 to issue "make clean" and "make". 
Next, I went to ~/Desktop/DynamixelSDK/c/example/protocol2.0/my_sync_read_write/linux64 to "make clean" and "make".

So, when the groupSyncWriteAddParam function is called, it should print out some information to help me to debug. This includes the statement to print on the screen that the program has entered the function. However, those print statements I added do not take effect. Did I make a mistake?

ROBOTIS-Will, 2020-09-15:
Hi Chun,

Thank you for your inquiries. Please see below information.

- Q10. `return 0` terminates the program. Looks like your program is getting an error during `groupSyncWriteAddParam`

- Q11. Make sure to reinstall the library with `sudo make clean && make reinstall` after modifying the library.
Check the [build library section](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/c_linux/#build-the-library) of the DYNAMIXEL SDK eManual.

- Q12. It is strongly recommended to test and learn how to use DYNAMIXEL with DYNAMIXEL Wizard 2.0 before actually getting into programming. You can test various features and functions to understand better when playing around with DYNAMIXEL Wizard 2.0 GUI environment.

- Q13. The Present Current is the actual current measurement that is flowing on the coil of the core motor, and this current can change very quickly due to the nature of the controller. We do not recommend to use this value to measure the precise Torque applied on DYNAMIXEL.

- Q14. Each DYNAMIXEL may have some differences which can cause minor discrepancy. Holding the horn will increase the Present Current value, but due to the gear ratio, the current may not reach up as high as you expect.

- Q15. If you are reading the value from your program, please cross check with DYNAMIXEL Wizard 2.0 to read the correct value. Your program may not correctly updating the read value.

- Q16. Negative value in Speed and Current stands for the opposite direction(CW).

robotiscustomer, 2020-09-15:
Hi Will, thank you for your reply.

Q13. What could be used to measure the precise Torque applied on the DYNAMIXEL?

Q15. In the read_write.c and sync_read_write.c sample programs, there is a do-while loop to check if the present position is close enough to the target position within the DXL_MOVING_STATUS_THRESHOLD. Once this condition is satisfied , the DYNAMIXEL moves to the next target position. I saved the position from the encoder, velocity and current which correspond to the time step when the last present position obtained before the do-while loop terminated. Is this the correct way? If not, please let me know the proper way.

Thank you.

ROBOTIS-Will, 2020-09-16:
Hi,

Q12. Please see [Graph](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#graph) section of DYNAMIXEL Wizard 2.0 manual.

Q14. In order to properly use the Present Current, you'll need some kind of filtering to get a meaningful data for your application. Present Current is accurate but will be more meaningful with proper data handling. On-Off for the robot gripper is one of the expected applications with Current-based Position control mode.

Q15. The example is given to help developers to understand how to use certain features by going over the source code. Without a thorough understanding about your application, I cannot determine whether the data acquisition timing and method in the example is correct for your case or not. If you are developing a product, you should have your software engineer look into the example code and make a suitable code to get the data from DYNAMIXEL for the application.

Thank you.

robotiscustomer, 2020-09-17:
Hi,

Thank you for your clarifications.

Q12. I am very sorry I missed it. Now I can see both the Present Velocity and Present Current curves. Thank you very much. I can see that the Present Current curve is much nicer looking than the noisy ones I got from the sync_read_write program. Is it because the displayed signals were filtered by the program before they were displayed under Graph in Dynamixel Wizard?

Q15: I just want to plot the velocity and current curves to see how they look like as the position changes over time. In this case, is my approach of saving these data using the approach in my previous post correct? I still cannot figure out why sometimes I got zero from Present Current. 

Thank you.

ROBOTIS-Will, 2021-01-29:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will