ROBOTIS-GIT/DynamixelSDK #434: Dynamixel XL 320

nitinsonker78, 2020-09-05:
ISSUE TEMPLATE ver. 1.1.1

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1], [3.5.2], etc.

- I'm using `C++` Language

  - [C] / [C++] / [C#] / [Python] / [Java] / [MATLAB] / [LABVIEW] / [ROS] / [Arduino]

- I'm using `USB2Dynamixel2` serial converter

  - [USB2Dynamixel] / [USB2Dynamixel2] or [OpenCM904] / [OpenCR] / [CM700s] / [CM500s] or other ROBOTIS product only

- I'm using `XM430-W210-R`

  - [MX-28], [AX-12W], [H54-200W], etc.

- and I'm having an `issue` like what

  - 'My motor doesn't work', etc.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

nitinsonker78, 2020-09-05:
can anybody help me with Dynamixel XL320  for joint mode to wheel mode with arduino uno

ROBOTIS-Will, 2020-09-11:
Hi @nitinsonker78 ,
In order to use Arduino Uno as your controller, you should use DYNAMIXEL2Arduino and DYNAMIXEL Shield library.
Please refer to the eManual below.
https://emanual.robotis.com/docs/en/software/arduino_ide/#library-api
Thank you.