ROBOTIS-GIT/DynamixelSDK #435: [3.7.31][Matlab][USB2Dynamixel2][XM430-W350-R] Issue : The called lib_name no mathed "groupSyncWrite"

liuqy1994, 2020-09-09:
ISSUE TEMPLATE ver. 1.1.1

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE WON'T FIX ANYWAY

STEP 1. Check what you are suffering from :

I'm using DynamixelSDK ver. 3.7.31

[3.7.31]
I'm using the Matlab Language

[Matlab]
I'm using U2D2 to TTL serial converter

[U2D2 to TTL]
I'm using U2D2

[U2D2]
and I'm having an issue like this:
`错误使用 calllib
未找到方法。

出错 groupSyncWrite (line 35)
VALUE = calllib(lib_name, 'groupSyncWrite', group_num, protocol_version,
start_address, data_length);

出错 sync_read_write (line 90)
groupwrite_num = groupSyncWrite(port_num, PROTOCOL_VERSION,
ADDR_PRO_GOAL_POSITION, LEN_PRO_GOAL_POSITION);`

STEP 2. Write Title as [3.7.31][Matlab][USB2Dynamixel2][XM430-W350-R] Issue : The called lib no mathed "groupSyncWrite"

Now, if you think :

Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of Title?

Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of Title?

or more that is not listed on above.

, You should write your issue on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

When I tried to run the example named `sync_write_read.m,` the following error occurred

I am try using `groupSyncWrite` in Matlab2018b on one Windows PC, But there are some error like this:
`错误使用 calllib
未找到方法。

出错 groupSyncWrite (line 35)
VALUE = calllib(lib_name, 'groupSyncWrite', group_num, protocol_version,
start_address, data_length);

出错 sync_read_write (line 90)
groupwrite_num = groupSyncWrite(port_num, PROTOCOL_VERSION,
ADDR_PRO_GOAL_POSITION, LEN_PRO_GOAL_POSITION);`

[image]

When i type `libfunction(lib_name)` to find What methods in the `lib_name`, then:
`>> libfunctions(lib_name)

类 lib.dxl_x64_c 的方法:

broadcastPing           pingGetModelNum         setBaudRate             
bulkReadTx              portHandler             setDataWrite            
bulkWriteTxOnly         read1ByteRx             setPacketTimeout        
clearMultiTurn          read1ByteTx             setPacketTimeoutMSec    
clearPort               read1ByteTxRx           setPortName             
closePort               read2ByteRx             syncReadTx              
factoryReset            read2ByteTx             syncWriteTxOnly         
getBaudRate             read2ByteTxRx           txPacket                
getBroadcastPingResult  read4ByteRx             txRxPacket              
getDataRead             read4ByteTx             write1ByteTxOnly        
getLastRxPacketError    read4ByteTxRx           write1ByteTxRx          
getLastTxRxResult       readPort                write2ByteTxOnly        
getPortName             readRx                  write2ByteTxRx          
getRxPacketError        readTx                  write4ByteTxOnly        
getTxRxResult           readTxRx                write4ByteTxRx          
isPacketTimeout         reboot                  writePort               
openPort                regWriteTxOnly          writeTxOnly             
packetHandler           regWriteTxRx            writeTxRx               
ping                    rxPacket    `

Obviously, there's no one method named **groupSyncWrite** here.

Any help will be thankful.

ROBOTIS-Will, 2020-09-17:
Hi,
Make sure that you have imported the function reference as shown below.
[https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/matlab_windows/#import-function-reference](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/matlab_windows/#import-function-reference)

liuqy1994, 2020-09-17:
@

> Hi,
> Make sure that you have imported the function reference as shown below.
> https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/matlab_windows/#import-function-reference

very glad to receive your reply！

ROBOTIS-Will, 2020-09-22:
Glad to hear it worked.
Thank you for your feedback.

closed 2020-09-22 by ROBOTIS-Will