ROBOTIS-GIT/DynamixelSDK #436: RX-64 present speed value more than 2047. It not real. How to fix ?

HongYu313Lin, 2020-10-15:
I need using RX-64 in wheel mode. 
I try to read speed, but When the position crosses the boundary like CCW from 1020 to 20 or CW 20 to 1000 and then speed value can be like 3154 or 4015.

I now E-manual show present speed is 0~2047 and 10th bit is direction, but It have two bytes. I can read more than 2047 about 3154 or 4015.....
Who knows what the other bits mean about 11th and 12th ? and how can I fixed ?  

I tried to observe the same situation in RoboPlus and DYNAMIXEL Wizard 2.0.

ROBOTIS-Will, 2020-11-03:
Hi @HongYu313Lin ,
RX series have a limited range of encoder reading therefore, when the horn crosses this "dead zone", speed value that is calculated based on the position value can be incorrect.
Thank you.

ROBOTIS-Will, 2021-01-29:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will