ROBOTIS-GIT/DynamixelSDK #438: Problem in setting up Dynamixel SDK under Ubuntu 20.10 Desktop for Pi

robotiscustomer, 2020-10-26:
Preliminaries:

I'm using U2D2 

I'm trying to set up Dynamixel SDK on a Raspberry Pi 4B running Ubuntu 20.10 Desktop for Pi. 
After executing 'sudo apt-get install build-essential' without issue, I came into an issue when setting up the library for c/c++ code (gcc version 10.2.0, Ubuntu 10.2.0-13ubuntu1) ) at the step:

$ sudo apt-get install gcc-multilib g++-multilib

It returns:
E: Package 'gcc-multilib' has no installation candidate
E: Unable to locate package g++-multilib
E: Couldn't find any package by regex 'g++-multilib'

From #301 , it looks like we can just ignore the error message so I proceeded to
'cd to /c/build/linux_sbc' and 'make'. Then, I got "make: *** [Makefile:68: livdxl_sbc_c.so] Error 1". Next, I typed "make clean && make" but the same error remains. 

Please let me know how to fix it. Thank you.

ROBOTIS-Will, 2020-11-03:
Hi @robotiscustomer 
Raspberry Pi 4 and Ubuntu 20.10 is a fairly new environment.
We'll review this issue and get back to you, but this may take a couple weeks.
Thank you.

robotiscustomer, 2020-11-03:
Hi @ROBOTIS-Will 
Thank you.

closed 2021-01-29 by ROBOTIS-Will