ROBOTIS-GIT/DynamixelSDK #439: Recommend baud rate for Dynamixel pro PM54-060-S250-R

metallo25, 2020-10-29:
- I'm using `DynamixelSDK ver.  3.7.31`

- I'm using `python` Language
 
- I'm using `U2D2` serial converter

- I'm using PM54-060-S250-R

I am trying to control two dynamixel pros in parallel, by having them on separate threads in python. They are connected to separate ports. 
Is it recommended to have it at baud rate of 1 Mb/second for this setup?. Because I am getting sometime out errors when doing so, when at 57600 per second. Please let me know your thoughts.

ROBOTIS-Will, 2020-11-03:
Hi @metallo25 
When increasing the baudrate, it is recommended to decrease the [Return Delay Time](https://emanual.robotis.com/docs/en/dxl/p/pm54-060-s250-r/#return-delay-time9) to minimize the delay.
You might also want to minimize the [USB port latency](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#usb-latency-setting) which is normally set to 16ms.
Thank you.

metallo25, 2020-11-03:
Ok, thanks for the input. for the dynamixel model mentioned above, it is good to be at 57600 baud rate or at 1 mb per second.
especially for parallel control of two dynamixel pro on different ports.

metallo25, 2020-11-12:
@ROBOTIS-Will : could you please reply above?

ROBOTIS-zerom, 2020-11-17:
@metallo25 : I think it's better to set it to 1 Mbps.
However, as @ROBOTIS-Will mentioned, it is recommended to change the latency timer value of all USB ports to 1ms.

ROBOTIS-Will, 2021-01-29:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will