ROBOTIS-GIT/DynamixelSDK #440: AX12-A ,present position return very high value(more than 6 digits)

mawin958700, 2020-10-30:
◉ I'm using DynamixelSDK ver. 3.7.31

◉ I'm using Python  Language

◉ I'm using USB2Dynamixel2 serial converter

◉ I'm using AX12-A

I've tried read_write.py (in test folder using protocol_1), so in emanual,  AX12 have limit position only 0-1023, but I got the value  
return from code as shown below

'Succeeded to open the port
Succeeded to change the baudrate
Dynamixel has been successfully connected
Press any key to continue! (or press ESC to quit!)
[ID:001] GoalPos:1023  PresPos:846
[ID:001] GoalPos:1023  PresPos:853
[ID:001] GoalPos:1023  PresPos:866
[ID:001] GoalPos:1023  PresPos:880
[ID:001] GoalPos:1023  PresPos:897
[ID:001] GoalPos:1023  PresPos:17302418
[ID:001] GoalPos:1023  PresPos:17302435
[ID:001] GoalPos:1023  PresPos:17302452
[ID:001] GoalPos:1023  PresPos:17302469
[ID:001] GoalPos:1023  PresPos:17302487
[ID:001] GoalPos:1023  PresPos:17302505
[ID:001] GoalPos:1023  PresPos:17302520
[ID:001] GoalPos:1023  PresPos:17302527
[ID:001] GoalPos:1023  PresPos:17302527
[ID:001] GoalPos:1023  PresPos:28574719
[ID:001] GoalPos:1023  PresPos:28574719
[ID:001] GoalPos:1023  PresPos:28574719
[ID:001] GoalPos:1023  PresPos:28574719
[ID:001] GoalPos:1023  PresPos:28574719
[ID:001] GoalPos:1023  PresPos:28574717
[ID:001] GoalPos:1023  PresPos:28574717
[ID:001] GoalPos:1023  PresPos:28574717
[ID:001] GoalPos:1023  PresPos:69731325
[ID:001] GoalPos:1023  PresPos:69731325
[ID:001] GoalPos:1023  PresPos:69731325
[ID:001] GoalPos:1023  PresPos:69731325
[ID:001] GoalPos:1023  PresPos:69731325
[ID:001] GoalPos:1023  PresPos:69731325
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:1023  PresPos:000
[ID:001] GoalPos:1023  PresPos:1021
Press any key to continue! (or press ESC to quit!)`

ROBOTIS-Will, 2020-11-04:
Hi @mawin958700 
Thank you for reporting the issue.
It seems like some trash values in the leading byte has not been cleared.
We'll review this and get back to you soon.

mawin958700, 2020-11-06:
@ROBOTIS-Will,  Thank you very much for your reply.

ROBOTIS-Will, 2020-11-12:
@mawin958700 
There is a bug in the example code that reads 4 byte position data while AX only use 2 byte position data.
Please use [read2ByteTxRx()](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol1_0/read_write.py#L127) and [write2ByteTxRx()](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol1_0/read_write.py#L119) for reading and writing Position values.
I'll fix this bug soon.
Thank you.

mawin958700, 2020-11-18:
Thank you very much, now I get  the good return value.

closed 2021-01-29 by ROBOTIS-Will