ROBOTIS-GIT/DynamixelSDK #441: sync_read_write.py is different from the one shown in the e-Manual

robotiscustomer, 2020-11-04:
Preliminaries: The code DynamixelSDK/python/tests/protocol2_0/sync_read_write.py that came with the SDK installation is different from the one listed and explained in the e-Manual.

In the e-Manual, they used:

`dxl_addparam_result = ctypes.c_ubyte(dynamixel.groupSyncWriteAddParam(groupwrite_num, DXL1_ID, dxl_goal_position[index], LEN_PRO_GOAL_POSITION)).value`

which is very similar to the way the C version is implemented.

In the actual program after installation of the SDK, there is a somewhat extra step:
```
# Allocate goal position value into byte array
param_goal_position = [DXL_LOBYTE(DXL_LOWORD(dxl_goal_position[index])),               DXL_HIBYTE(DXL_LOWORD(dxl_goal_position[index])), DXL_LOBYTE(DXL_HIWORD(dxl_goal_position[index])), DXL_HIBYTE(DXL_HIWORD(dxl_goal_position[index]))]

# Add Dynamixel#1 goal position value to the Syncwrite parameter storage
dxl_addparam_result = groupSyncWrite.addParam(DXL1_ID, param_goal_position)
```

Can you tell me why the approach shown in the e-Manual has been replaced? What is the reason for using the extra step in the actual Python code? It seems to reverse the order of the bytes that are used as 32-bit representation of the positional value to match the representation of address in Protocol 2.0. Thank you.

ROBOTIS-David, 2020-11-16:
Hi, RobotisCustomoer,

It seems that the e-Manual is not updated since the code is changed. 

I think it is better to use the current distributed code. 

Below is the test that I carried out using the current distributed code, which work okay. 
[image]

Regarding the reversing-ordered bytes ("HIBYTE", "LOBYTE"), the parameters can be stored Low < High according to the Little Endian law. For your reference, refer to the [Instruction Packet](https://emanual.robotis.com/docs/en/dxl/protocol2/#instruction-packet), [Sync Write](https://emanual.robotis.com/docs/en/dxl/protocol2/#sync-write). A table in the page indicates how the packet is transmitted to the DYNAMIXEL. 

Thank you,

robotiscustomer, 2020-11-17:
Hi @ROBOTIS-David 

Thank you

closed 2021-01-29 by ROBOTIS-Will