ROBOTIS-GIT/DynamixelSDK #442: [3.7.31][Python3][XH540-W270R] Issue: Sometimes Present Current returns very large values

robotiscustomer, 2020-11-04:
I am using DynamixelSDK ver. 3.7.31

I am using Python3 Language

I am using U2D2

I put a Dynamixel on my desk and there is no load attached to the horn. Sometimes Present Current returns very large values. For example, after multiplying the Present Current by 2.69 mA, I got:

2.7 
21.5 
10.8 
24.2 
0.0 
176286.5 
176259.6 
176259.6 
176275.7 
176264.9 
176254.2 
176267.6 
176267.6 

Tried another Dynamixel and also got very large values sometimes.

ROBOTIS-David, 2020-11-16:
Hi, robotiscustomer, 

Seems that you tried reading the 4 bytes consecutive data from Present Current address to the next.

Note that the _Present Current_ data, not like 4 bytes size _Present Position_, occupies 2 bytes size in the control table, so that 2 byte reading method should be used.  You can find more about API reference to Python code for SDK. See [Python PacketHandler](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/api_reference/python/python_packethandler/#python-packethandler)

I simply tested _read_write.py_ file to get _Present Current_ data by modifying my code (I used [_read2ByteTxRx_ ](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/api_reference/python/python_packethandler/#read2bytetxrx)), it results in following (Present Current in Position Mode).

[image]

You may wonder why there is a huge value present in the output command. It is because DYNAMIXEL defines the negative value with Two's complement. For more information on this, refer to [a previous issue ticket](https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/264) that had been already addressed. 

Thank you, 

Thank you,

robotiscustomer, 2020-11-19:
Hi @ROBOTIS-David 

Thank you for mentioning about the Two's complement issue and providing a solution. Now the values became more reasonable.  Thank you.

ROBOTIS-Will, 2021-01-29:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will