ROBOTIS-GIT/DynamixelSDK #444: [TxRxResult] There is no status packet! MX-64 Dynamixel Servomotor

Olgarubio, 2020-11-18:
ISSUE TEMPLATE ver. 1.1.1

Hello, I'm new programming on python, I'm driving 7 motors with the U2D2 by Dynamixel, I have 4 AX-12 series Motors, 2 AX-18A motor and 1 Mx-64 T.... I have an issue with the communication with the MX series. 

Below my code ..

from dxl_control.Ax12 import Ax12

#import BlynkLib
#import time

# create motor object

my_dxl_7 = Ax12(7)
my_dxl_6 = Ax12(6)
my_dxl_5 = Ax12(5)
my_dxl_4 = Ax12(4)
my_dxl_3 = Ax12(3)
my_dxl_2 = Ax12(2)
my_dxl_1 = Ax12(1)

bool_test = True
pos1_robot=[0,0,0,0,0,0,0]
pos2_robot = [0,0,0,0,0,0,0]
pos3_robot=[0,0,0,0,0,0,0]
pos4_robot=[0,0,0,0,0,0,0]
pos_edit = 0
pos_robot=[0,0,0,0,0,0,0]
pos_return=0
pos_test=0

# connecting

Ax12.open_port()

Ax12.set_baudrate()
#blynk = BlynkLib.Blynk('Ralox0Jlha2liIjodCme2FONyYIXb05p')

def user_input():

    """ Check to see if user wants to continue """

    ans = input('Desesa moverlo nuevamente?')

    if ans == 'n':

        return False

    else:
      
        return True

def test_pos(motor_object,pos_edit):

    bool_test = True

    while bool_test:

        motor_object.get_position()

        # desired angle input

        input_pos = int(input("input pos: "))

        motor_object.set_position(input_pos)

        bool_test = user_input()

        pos_test = 150

        return pos_edit

def mov_new_pos(motor_object,new_pos):

   

        motor_object.get_position()

        motor_object.set_position(new_pos)

  

def Modificar():

    """ Check to see if user wants to continue """

    mover_eje = int(input('Que eje quieres mover?'))

    mod_pos_robot = input('Que posicion deseas modificar?')

    if mover_eje == 0:

        test_pos(my_dxl_1,pos_return)

    if mover_eje == 1:

        test_pos(my_dxl_2,pos_return)
    if mover_eje == 2:

        test_pos(my_dxl_3,pos_return)
    if mover_eje == 3:

        test_pos(my_dxl_4,pos_return)
    if mover_eje == 4:

        test_pos(my_dxl_5,pos_return)
    if mover_eje == 5:

        test_pos(my_dxl_6,pos_return)
    if mover_eje == 6:

        test_pos(my_dxl_7,pos_return)

    guarda_pos = input('Desea guardar posicion? ')

    if guarda_pos == 'y':

        if mod_pos_robot == '1':

            pos1_robot [0] = my_dxl_1.get_position()
            pos1_robot [1] = my_dxl_2.get_position()
            pos1_robot [2] = my_dxl_3.get_position()
            pos1_robot [3] = my_dxl_4.get_position()
            pos1_robot [4] = my_dxl_5.get_position()
            pos1_robot [5] = my_dxl_6.get_position()
            pos1_robot [6] = my_dxl_7.get_position()

        if mod_pos_robot == '2':

            pos2_robot [0] = my_dxl_1.get_position()
            pos2_robot [1] = my_dxl_2.get_position()
            pos2_robot [2] = my_dxl_3.get_position()
            pos2_robot [3] = my_dxl_4.get_position()
            pos2_robot [4] = my_dxl_5.get_position()
            pos2_robot [5] = my_dxl_6.get_position()
            pos2_robot [6] = my_dxl_7.get_position()
        
        if mod_pos_robot == '3':

            pos3_robot [0] = my_dxl_1.get_position()
            pos3_robot [1] = my_dxl_2.get_position()
            pos3_robot [2] = my_dxl_3.get_position()
            pos3_robot [3] = my_dxl_4.get_position()
            pos3_robot [4] = my_dxl_5.get_position()
            pos3_robot [5] = my_dxl_6.get_position()
            pos3_robot [6] = my_dxl_7.get_position()

        if mod_pos_robot == '4':

            pos4_robot [0] = my_dxl_1.get_position()
            pos4_robot [1] = my_dxl_2.get_position()
            pos4_robot [2] = my_dxl_3.get_position()
            pos4_robot [3] = my_dxl_4.get_position()
            pos4_robot [4] = my_dxl_5.get_position()
            pos4_robot [5] = my_dxl_6.get_position()
            pos4_robot [6] = my_dxl_7.get_position()

    print('posicion guardada')

def posicion():

    Auto_pos = input('A que posicion desea ir? ')

    if Auto_pos == '1':

        pos_robot = pos1_robot 

    if Auto_pos == '2':

        pos_robot = pos2_robot

    mov_new_pos(my_dxl_1,pos_robot[0])
    mov_new_pos(my_dxl_2,pos_robot[1])
    mov_new_pos(my_dxl_3,pos_robot[2])
    mov_new_pos(my_dxl_4,pos_robot[3])
    mov_new_pos(my_dxl_5,pos_robot[4])
    mov_new_pos(my_dxl_6,pos_robot[5])
    mov_new_pos(my_dxl_7,pos_robot[6])

while True:

    print('seleccione una opcion')

    print('p= ir a posicion')

    print('m= modificar posicion')

    print ('s= salir')

   

    opcion = input('ingrese la opcion deseada:')

    if opcion == 'p':

        posicion()

       #posicion()

    if opcion == 'm':

        

        Modificar()

    

    if opcion == 's':

    

        break

#test_pos(my_dxl)

#test_pos(my_dxl_2)

# disconnect

my_dxl.disable_torque()

Ax12.close_port()

The error is :

[TxRxResult] There is no status packet!

[image]

BE CAREFUL!! FOLLOW THE RULES AS FOLLOWS, OR YOUR ISSUE WILL BE `WON'T FIX` ANYWAY

STEP 1. Check what you are suffering from :

- I'm using `DynamixelSDK ver. 3.5.4`

  - [3.4.1], [3.5.2], etc.

- I'm using `C++` Language

  - [C] / [C++] / [C#] / [Python] / [Java] / [MATLAB] / [LABVIEW] / [ROS] / [Arduino]

- I'm using `USB2Dynamixel2` serial converter

  - [USB2Dynamixel] / [USB2Dynamixel2] or [OpenCM904] / [OpenCR] / [CM700s] / [CM500s] or other ROBOTIS product only

- I'm using `XM430-W210-R`

  - [MX-28], [AX-12W], [H54-200W], etc.

- and I'm having an `issue` like what

  - 'My motor doesn't work', etc.

STEP 2. Write `Title` as `[3.5.4][C++][USB2Dynamixel2][XM430-W210-R] Issue : My motor doesn't work`

- Now, if you think :

  - Hey, I'm using RoboPlus2.0, and having an issue, but how can I write that kind of `Title`?

  - Hey, I'm using USB2AX serial converter, and having an issue, but how can I write that kind of `Title`?

  - or more that is not listed on above.

- , You should write your `issue` on http://en.robotis.com/BlueAD/board.php?bbs_id=old_qna

STEP 3. Delete all written here, and describe what your problem is

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)

ROBOTIS-Will, 2020-11-24:
Hi @Olgarubio 
For me, it looks like you are not using DYNAMIXEL SDK, but some other 3rd party library which I cannot help you with.
I recommend you to try running below DYNAMIXEL SDK example, but you'll need to modify the address definitions depending on your [MX-64](https://emanual.robotis.com/docs/en/dxl/mx/mx-64/#control-table-of-eeprom-area).

https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol1_0/read_write.py
Thanks

ROBOTIS-Will, 2021-01-29:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-01-29 by ROBOTIS-Will