ROBOTIS-GIT/DynamixelSDK #454: [3.7.31][Matlab][USB2Dynamixel2][XM430-W350-R] Torque mode control performance is poor

liuqy1994, 2020-11-26:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) `3.7.31`

2. Which programming language/tool do you use?
  - ex)  `MATLAB`,

3. Which operating system do you use?
  - ex) `Windows 10`

3. Which USB serial converter do you use?
  - ex)  `U2D2`

4. Which DYNAMIXEL do you use?
  - ex) `XM430-W350-R`

5. Have you searched the issue from the closed issue threads?
`Yes`
6. Please describe the issue in detail
I am using the torque control mode to control the OpenManipulator through matlab,  I designed a simple PID control structure referring to the current control mode of the [e-manual.](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/)
In each control loop, I read the angle and angular velocity of the five servos (sync read), and write the current value of each servo through the calculation of pid (sync write). 
However, the joint motion of the robotic arm is not silky at all, and it appears as a step like the picture below (joint-13).
[image]
**Is this the reason for the low control frequency?**
I measured the actual control frequency through matlab's tic and toc to be about 50Hz. It takes about 5 milliseconds to read or write data once. Any help to improve performance will be appreciated！ @ROBOTIS-Will
7. How can we reproduce the issue?
If necessary, I can upload the code.

treepeony, 2020-12-04:
I'm just starting out with openmanipulator/Matlab as well. What baud rate are you using to communicate?

liuqy1994, 2020-12-05:
> I'm just starting out with openmanipulator/Matlab as well. What baud rate are you using to communicate?

Hi, The baud rate is 4M in my experiment.

ROBOTIS-Will, 2021-01-04:
@liuqy1994 
Hi, I'm sorry about the delayed response on this.
50Hz(20ms) frequency seems too slow. Have you check the USB port latency which is set to 16ms by default?
You can refer to the [Graph Optimization section of DYNAMIXEL Wizard 2.0](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#graph-optimization) which describes how to decrease the port latency.
Thank you.

liuqy1994, 2021-01-06:
@ROBOTIS-Will 
Hi, thank you for your reply.

I have modified the USB port latency to 1ms and setted the Return Delay Time(9) to zero(0).
Now the frequency of reading data can reach 100Hz.
But when the data is read more frequently, the value obtained is a large deviation from the previous data or even 0.
So, is there any good way to solve this problem?
What is the maximum frequency at which the servo can be read ?

Looking forward to your reply.

Thank you very much!

ROBOTIS-Will, 2021-01-07:
@liuqy1994 
Could you give us more detail about which data you are reading from DYNAMIXEL?
If possible, please also check the actual time that your software transmits the sync read instruction.
An optimized graph of DYNAMIXEL Wizard 2.0 reads data at 1ms(1000Hz) frequency, but this may vary depending on the number of DYNAMIXEL and data you read.
Thanks.

liuqy1994, 2021-01-07:
@ROBOTIS-Will 
The data I tried to read include position, speed and current from OpenManipulatorX with 5 servos, using Matlab language and U2D2 adapter board.
I tried three modes, normal, synchronous and indirect address mode. The fastest frequency measured using the third mode(indirect address mode) is 100Hz.
If it takes 1 millisecond for each servo to read data, then it takes 5 milliseconds to read the data of the robotic arm. In other words, is the highest frequency of data transmission of the robotic arm 200 Hz?

Thanks a lot.

ROBOTIS-Will, 2021-01-07:
Oh, when you are accessing identical set of addresses of multiple DYNAMIXELs, it is recommended to use Sync Read / Write which can reduce maximum 80% of communication delay.
Please refer to below issue thread.
https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/320
Thanks.

liuqy1994, 2021-01-07:
Sync Read / Write is more faster than Indirect address  Read / Write mode?
So, if I understand correctly, in my case, when I try to read three kinds of data(position, speed and current)  at the same time, the Sync Read / Write mode is the most recommended.
But, When I try to increase the reading frequency, the following dxl_getdata_result will return a **False**.
`dxl_getdata_result = groupSyncReadIsAvailable(groupread_num, DXL1_ID, ADDR_PRO_PRESENT_POSITION, LEN_PRO_PRESENT_POSITION);`
Why does the servo return a false? And how to fix this?

Thanks again.

ROBOTIS-Will, 2021-01-07:
In the issue thread that I linked, there's a description about limitation of Sync Read/Write.

> For your case, Sync Read and Sync Write will be the feasible solution. However, in order to use them, you have to meet certain conditions; data has to be sequentially located in the memory address.

If you are only reading Goal Position from multiple DYNAMIXELs, you don't need to use the indirect address/data.

The example code is written for DYNAMIXEL P series, so you have to modify several definitions(such as Control Table Adress, Default Settings) so that the code can communicate with X series.
Thanks.

liuqy1994, 2021-01-07:
Okay, I’ll test it in an experiment to see if the frequency of reading data has increased.

Thank you for your time.

yun-goon, 2025-02-13:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2025-02-13 by yun-goon